QIDITECH/QIDI_MAX3 #1: The system starts abnormally when starting X-Max 3

ImNanou, 2023-08-15:
Hello, 

Just got an issue with my X-Max 3 printer, when I start it it says "The system starts abnormally. Please turn off the printer and wait for a while before restarting".

Is there a way to fix it or hard reset the mks board ?

Thanks in advance !

QIDITECH, 2023-08-26:
I'm very sorry that I couldn't reply you in time. Next week, we will arrange people to reply to the issues on github every day.

Please contact our after-sales mailbox, and provide your specific operation with a video or picture, so that we can determine the problem.

X-Max 3: Max3support@qd3Dprinter.com / Max3AMS@qd3Dprinter.com

buulke, 2023-09-17:
Mine is solved after reboot key and several times trying to update, it succeeded.
reboot key i got from service department online.

closed 2023-09-22 by ImNanou