QIDITECH/QIDI_MAX3 #2: Resume mode after turn off

Booldegum, 2023-09-01:
Hello, 
Will you add a resume function after cancel a print or after a power out?  I mean as on X-Plus V1 but " more efficient" (with more than one print resume for those who have several plates for example). This should be very useful for long prints

QIDITECH, 2023-09-02:
Sorry, X-MAX3 can not resume after power off.
Because it uses the Z-axis stop on the nozzle, which means that after power off, the Z-axis will not return to zero.

closed 2023-09-02 by QIDITECH

igoutta, 2024-04-17:
> Sorry, X-MAX3 can not resume after power off. Because it uses the Z-axis stop on the nozzle, which means that after power off, the Z-axis will not return to zero.

That's a terrible feature. Recently, our company bought your printer biased in the printing speed, but the lab had their resentments. Now we confirm our reservations about the quality of your products, since it's need to resume the printing process after continuous power outbreaks in the city.

Fisheye3D, 2024-04-17:
> That's a terrible feature. Recently, our company bought your printer biased in the printing speed, but the lab had their resentments. Now we confirm our reservations about the quality of your [...]

So printer that uses klipper firmware supports power loss recovery they see it as inaccurate and not reliable, Any company would be using a UPS anyhow.