QIDITECH/QIDI_MAX3 #3: Z axis inverted since update

Booldegum, 2023-09-14:
Hello, 
I just finish the firmware update for 4.3.8 and when i try to do the leveling i get "Endstop a still triggered after retract". After researches in the dashboard i finally find that Z axis seems inverted. So my printer is currently unusable... What to do?

I shouldn't have updated my printer

buulke, 2023-09-15:
Hi  
i have thesame problem, please fix this Qidi..   there still a few bugs in it.

buulke, 2023-09-17:
Mine is solved after reboot key and several times trying to update, it succeeded.
reboot key i got from service department online.

Rainboooom, 2023-09-25:
I'm very sorry, this is my mistake. The Max model has two different probe configurations, and I have included them in the upgrade package, but I haven't provided a specific operation process. Now I don't have the specific operation process listed in the README.me document. Please follow this process to operate. I'm very sorry for the bad experience you had.

buulke, 2023-09-25:
Hi
a whole lot of rebooting and software updating it does it again, sometimes it doesn't start the print or makes the work piece very weird.

the system also crashes regularly

Regards Mart

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows

Rainboooom, 2023-09-25:
This is a bit strange because the upgrade package will overwrite the files from the previous version. You can contact our after-sales support team to provide us with specific feedback, and we will handle these as soon as possible.
After sales website: https://qidi3d.com/pages/warranty-policy-after-sales-support

closed 2023-09-27 by Rainboooom