QIDITECH/QIDI_MAX3 #4: Nach Update auf 4.3.8 rammt Nozzle ins Bett

rt1970, 2023-09-16:
Nach Update auf 4.3.8 und trotz mehrfach neu Kalibrieren von Mesh und Z-Probe hat der Drucker immer wieder meine Nozzle ins Druckbett gerammt!
Nachdem mein Druckbett und meine Nozzle ZERSTÖRT ist, habe ich nun heraus gefunden, dass es Messfehler beim Z-Probe gibt!
Ich hatte einen Messfehler von 2mm!
Z-Probe wurde mit 3,7 ermittelt!
Nachdem ich den Z-Probe OHNE ZU MESSEN auf 1.7 gesetzt habe druckt er endlich wieder!
Mal schauen was der Support sagt...

rt1970, 2023-09-18:
Lösung ist: falsche Firmware geladen (obwohl ich mir 100% sicher war die für BL-Touch rechts geladen zu haben!)
Nach ersetzen der printer.cfg funktioniert wieder alles!
Natürlich musste ich die Nozzle ersetzen und das Druckbett muss auch noch neu!!!

Rainboooom, 2023-09-25:
Es tut mir sehr leid, ich habe wichtige Details über die Auswahl verschiedener Konfigurationsdateien für verschiedene Sonden beim Hochladen der Upgrade-Paketdatei verpasst. Leider wurde Ihr Drucker aufgrund meines Fehlers beschädigt, und es tut mir auch sehr leid, dass mein Fehler Ihnen eine schlechte Erfahrung bereitet hat. Es tut mir sehr leid, aber bitte kontaktieren Sie unseren Kundendienst über diese Website: https://qidi3d.com/pages/warranty-policy-after-sales-support

Entsprechend haben wir auch die neuen Betriebsabläufe aktualisiert.

---

I'm very sorry, I missed important details about choosing different configuration files for different probes when uploading the upgrade package file. Unfortunately, your printer was damaged due to my mistake, and I'm also very sorry that my mistake gave you a bad experience. I'm very sorry, but please contact our after-sales support through this website: https://qidi3d.com/pages/warranty-policy-after-sales-support

Correspondingly, we have also updated the new operating procedures.

closed 2023-09-25 by Rainboooom