QIDITECH/QIDI_MAX3 #5: Cannot rename disk for firmware update to ' ' with Linux

uhthomas, 2023-10-04:
The guide for updating the firmware wants the USB disk to be renamed to ' ', but this does not seem possible with Linux.

[image]

See:

[image]

Is it possible to allow other names, which will be compatible across platforms?

uhthomas, 2023-10-04:
`mlabel` is the same.

```sh
❯ sudo mlabel -i /dev/sda1 ::" "
Long file name " " is reserved.
a)utorename A)utorename-all r)ename R)ename-all
s)kip S)kip-all q)uit (aArRsSq): 
``

Rainboooom, 2023-10-05:
You can now update the USB device without renaming it to a blank name.

In previous versions, some machines had some issues with USB device recognition, so we suggest setting your USB device name to blank. This issue no longer exists, so there is no need to rename the USB device again. Thank you for your feedback, which made me aware of the missing updates in the corresponding process. I will modify the update process together in the next update, We are very sorry for the inconvenience caused to you, and we also appreciate your feedback for making progress.

closed 2023-10-09 by Rainboooom