QIDITECH/QIDI_MAX3 #6: mcu 'mcu': Unable to connect

rt1970, 2023-10-10:
After i try to upgrade the MCU´s i think my STM32F402 Firmware is defective...
Also my printhead-board.
My printhead-board i have:
./kiauh/kiauh.sh -> advanced -> Build only -> RP2040 -> flash chip generic_03H -> USB -> USB ids = D5CE4D9543D14858 (Quit and save)
Then i have hold „Boot“ und press „Reset“ on printhead-board.
After that i have cd /out/ -> cp klipper.uf2 /home/mks/gcode-files/sda1/
Now it show the device with ls /dev/serial/by-id
usb-Klipper_rp2040_D5CE4D9543D14858-if00

But what is the config in kiauh for STM32F402? Whats the name for klipper.bin on micro-sd or can i programming with STM32CubeProgrammer in DFU-Mode?

Rainboooom, 2023-10-14:
Please contact our professional after-sales support as soon as possible. In order to cooperate with our other basic communication, the direct communication between the motherboard and adapter board is specially configured.

rt1970, 2023-10-18:
I have contact Qidi-Support. A new Mainboard is now on the way to me!
THANK YOU QIDI for your professional support!
Btw:
I'm really excited about the QIDI X-MAX 3 and the support!!! It is my best Printer!
Now I'm waiting for camera shipment...

Rainboooom, 2023-10-19:
Thank you for your support. By the way, you can change the default resolution of the camera yourself. Currently, the most conservative default settings should be used for the factory cameras, but the camera can support up to 1920 * 1080, 30fps
Simply connect to the fluid page and select webcam.txt in the configuration bar, as shown in the following figure to make modifications
[image]

rt1970, 2023-10-19:
Thank you very much. But the Qidi Cam is in pre-order.
I tried an old webcam. Then I found out that my cam is not a full HD cam.
My changes did not take effect. After that I edit webcam.txt from 1920x1080 to 1280x720. Now I have a 720p live stream.
Then I installed Timelaps and edited GCODE TIMELAPSE_TAKE_FRAME in the QIDI slicer when changing layers. Now I have a perfect time-lapse video. (like my old Octoprint printer)
I'm very satisfied!

rt1970, 2023-10-20:
Yeah! The new Mainboard was comming and my Printer is like new!
One Question: i think from time to time or restart (?) is the filament-detector in Klipper off.
What im doing wrong?

Rainboooom, 2023-10-25:
I'm very sorry for taking so long to reply to you. I've been quite busy these days.

Generally speaking, the opening or closing of the material cut-off switch does not affect the system restart. It is difficult for me to understand the specific situation to confirm whether the input voltage is stable. You can contact our professional after-sales team for support.

Nazrac, 2023-10-26:
So I did change my camera settings, the resolution changed, but no matter what its stuck at 15fps and the camera is capable of higher and my network is plenty fast enough. Not sure why its locked at 15fps

Rainboooom, 2023-10-27:
[image]
[image]
Try modifying these two areas. The camera hardware specification supports up to 30FPS

closed 2023-11-20 by QIDITECH