QIDITECH/QIDI_MAX3 #7: Park location is wrong

Nazrac, 2023-10-19:
Park location is at the max on the Y axis (325) one this is not ideal location for doing a filament change or whatever, but also on my machine the max Y value is beyond the mechanical limits of the machine. This may not be the case on others machines, but that is how it is on mine and after a pause the home is off because its skipped steps. Ideally Park should be at an easy location so you can reload purge and remove the tail before starting the print. I have moved mine to X5 Y5 and left the Z unchanged Still it appears the home values are not correct for Y, I will fix it in my firmware, but not everyone using this machine is as familiar with the CFG files and it should not be this way. There are other choices made in the config that are questionable at best.

Rainboooom, 2023-10-25:
It's like this, the pause position is at the extreme position of the Y-axis because in most printing situations, the printing space used is the central part, and the machine needs some movement in the design of the front panel edge to avoid friction with the edge. What you said is very reasonable, and we may come up with a more reliable configuration plan in future versions to optimize these issues.

Nazrac, 2023-10-26:
I already changed mine, but since you move the Z axis down it really doesn't matter where you park the head. I have placed it at 5,5 and at this point unless there is a print directly at this point it wont purge onto the print, but as I said even at max/2,max X,Y relative its still over the possible print location, not to mention with the new cover with your settings that wasn't even possible as the cover hit the rear of the machine by 2 mm throwing the home off when it skipped steps. Forget the Pause location as thats all opinion, how about fixing the fact that it couldn't even reach 325 in the Y axis as it was beyond the movement capabilities of the machine. Here is my printer.cfg file you can let me do some of your job for you. 
[printer mod.zip](https://github.com/QIDITECH/QIDI_MAX3/files/13178547/printer.mod.zip)

Rainboooom, 2023-10-30:
We will use this configuration for testing, and thank you very much for your suggestion.

closed 2023-11-20 by QIDITECH