QIDITECH/QIDI_MAX3 #10: Add unload button to pause menu

Nazrac, 2023-11-17:
I have had a few jams due to manually unloading filament using the normal buttons, perhaps we can add an unload button to run that routine from the pause menu, would make multi color prints much easier

QIDITECH, 2023-11-20:
Thank you for your feedback, we will consider your suggestions and optimize the UI.

tome9111991, 2023-12-07:
Hello.
On firmware .12
Replace filament after runout in a print annoying.
You have to open the ptfe tube on toolhead to cut filament.
I dont like to open the door if im printing ABS etc..

It is easier to open the ptfe after runout sensor.
[image]

After open the tube you can retract old filament.

QIDITECH, 2023-12-07:
Because many users were stuck when they paused and unloaded consumables, we canceled the return of consumables. If you prefer to return material, you can modify the M603 code in print.cfg via fluidd.

tome9111991, 2023-12-07:
> Because many users were stuck when they paused and unloaded consumables, we canceled the return of consumables. If you prefer to return material, you can modify the M603 code in print.cfg via fluidd.

Right i can modify the M603
But after a firmware update a have to modify the printer.cfg again

But maybe a extra Button to unload the filament would be nice.
Then the user can decide for himself how he wants to replace/unload the filament.

coderm4ster, 2023-12-18:
> But maybe a extra Button to unload the filament would be nice. Then the user can decide for himself how he wants to replace/unload the filament.

This would be very useful!

Cutting the filament above the head is very frustrating. Please bring back the unload button!

next-comp, 2024-03-12:
Another vote - please bring back the Unload button !!! 

I cannot upgrade my X-Plus 3 to 4.2.13 (from 4.2.10) because of the way you changed the unload of the filament. With 4.2.10 the Unload works flawlessly each time used. 4.2.13 is a downgrade not an upgrade because of that.

This became problem for many users. Please read: https://forum.drucktipps3d.de/forum/thread/33464-filament-entladen-und-laden-mit-neuer-firmware/   (use Google Chrome to translate to English).

Phil1988, 2025-01-25:
@tome9111991 @coderm4ster @next-comp @freneter 
Something like this?
[image]