QIDITECH/QIDI_MAX3 #12: Only Black Thumbnails

shubukin, 2023-11-27:
there are only black thumbnils when using orca or superslicer.

how can ich switch to stock klipperscreen or fix this issue

QIDITECH, 2023-11-28:
Sorry, this function is not currently supported, please look forward to the update in the future version.

Phil1988, 2023-12-01:
I read that thumbnails being visible when only use "upload" gcode not when using "upload & print".
Does this fail on both ways @shubukin ?

QIDITECH, 2023-12-04:
It is not caused by these reasons.
It's because the encoding formats of thumbnails are different, and our thumbnail format is not available in orca and superslicer.
We will be compatible with the formats they use in subsequent versions, but the loading speed may be slower.

tome9111991, 2023-12-07:
Another option would be to have your slicer dev team create a pull request in Orca/Superslicer git to support your thumbnail format.

Phil1988, 2023-12-13:
I dint have thumbnails at all even when using Qidi Slicer (1.0.7 and 1.0.8).

It worked once, but I dont know why not any more:
[image]
(everything from 2022 was "pre installed" and not sliced by me)

QIDITECH, 2023-12-14:
The thumbnail is selected as QIDI, can be displayed on the touch screen, but fluidd cannot be displayed.
The thumbnail is selected as PNG, which cannot be displayed on the touch screen, but fluidd can be displayed.
The next version of the firmware will be compatible with PNG format and will switch all formats to PNG.

Phil1988, 2024-01-06:
yeah its better if QIDI implements the PNG to the printer, so not every slicer has to change to support these 3 printers from QIDI.

BTW: the response from you @QIDITECH is not true.
Currently the PNG export of QIDI Slicer does not work correctly.
So choosing "QIDI" as image format will work on the display, but not on fluidd.
Chossing "PNG" will NOT work on fliudd and also NOT work on the display.

This is still an issue, even you closed it here:
https://github.com/QIDITECH/QIDISlicer/issues/17

Hope to see the new release soon which will fix this :)

QIDITECH, 2024-01-08:
Due to hardware constraints, PNG files are very slow to load, and we are still trying to solve this problem.
And the bug on QIDIslicer, we have fixed.

potens1, 2024-01-09:
I may say stupid things here, but the limited hardware is the screen, right ? Then can't you do something like when klipper is sending the gcode to it, it "updates" the preview using the main board CPU since it has way more power ?

QIDITECH, 2024-01-10:
Orca already supports ColPic thumbnails.