QIDITECH/QIDI_MAX3 #14: Bug in macro M603 with v4.3.12

fran6p, 2023-12-02:
The last version, v4.3.12 has a bug in the macro M603.
As is, the filament isn't retracted but pushed.
The correction is easy: change G0 E100 F400 by G0 **E-100** F400

By the way, there are inconsistencies in the last sentences in the description of the last release : 
 
    SOC_Version:V4.1.12
    Matching UI version：V4.1.12
    Please download MAX_V4.1.12.rar. It's packaged version.

 Shouldn't it be v4.**3**.12 instead of 4.1.12

closed 2023-12-04 by QIDITECH

tome9111991, 2023-12-19:
> Pausing and returning the material during printing can easily cause filament clogging. Please cut off the filament and then feed it.

Btw.
On old firmware like .10
The issue with clogging during unload
Was the M603 by default you extrude filament before unload `G0  E15 F400` just remove that and you don't have clogging during unload

coderm4ster, 2023-12-19:
@QIDITECH The approach to cutting the filament is catastrophic. Anyone who frequently changes filaments will always remove the top of the printer, pull out the tube, cut it with pliers, and then put everything back. This is not realistic but rather ridiculous and unprofessional.