QIDITECH/QIDI_MAX3 #17: Please bring back the unload button!

coderm4ster, 2023-12-19:
I just updated to firmware version 4.3.12 from 4.3.8.

I find the new filament change button to be not useful at all, rather quite annoying. Did someone at Qidi seriously think that, for every filament change, users would take off the printer's top, pull out the tube, cut the filament, and repeat the process in reverse just because it occasionally clogs for some users? This is disastrous, ridiculous, and unprofessional.

Moreover, for some reason, I can't even pull out the tube from the head, it got stuck (successfully frustrated me). Clogging rarely occurred for me, and even when it did, it was during printing simply because there was no unload button to advance the filament a bit before retracting.

Please bring back the unload button in the 'Load' menu and during printing in the 'Pause' menu! Thank you!

tome9111991, 2023-12-19:
I can confirm, filament change after runout frustrating.
If you print ASA/ABS or etc.
you have to open the front door the problem is that the chamber cools down because cold air comes in.
this leads to warping.

it is inconvenient to cut the filament.
it would be easier to open the PTFE after the filament runout sensor.

but since some users seem to have problems with clogging, i can say that removing `G0 E15 F400` helped me.
i use only  `G0 E-100 F300` to unload filament works great for me.

QIDITECH, 2023-12-20:
OK，We will add unload button back in the next version.

closed 2023-12-20 by QIDITECH

coderm4ster, 2023-12-20:
@QIDITECH Thanks a lot!!!

I love this printer, but the removal of the unload button and functionality has greatly frustrated me.
I apologize if I expressed it too drastically.