QIDITECH/QIDI_MAX3 #19: Please provide an .stl of the printbed

Phil1988, 2023-12-30:
It would be nice if you could share the .stl file of the print bed, so we can import that to our favorite slicer.

on QidiSlicer it looks like this:
[image]

while on PrusaSlicer the printbed looks like this:
[image]

if you could share the .stl file we will all have the nice looking printbed on our slicers.

Thanks in advance :)

andy775, 2023-12-31:
Phil,
[QIDISlicer_bed_stls.zip](https://github.com/QIDITECH/QIDI_MAX3/files/13802090/QIDISlicer_bed_stls.zip)

STL of bed images can be found in the installation directory of qidi slicer. I've attached the stls found in the path. 

Check the following path:

\\Program Files\QIDISlicer\resources\profiles\QIDITechnology

Happy New Year :)

Phil1988, 2024-01-01:
Thanks @andy775 that helped a lot ^^

closed 2024-01-01 by Phil1988