QIDITECH/QIDI_MAX3 #21: Feature request - Current Print LAyer

pmbroth, 2024-01-12:
Is it possible that on the printer., when printing, it displays current layer being printed, and total layers.   Example  Curerent Layer 23  Total Layers 600.   I have had this on other printers, and it was useful.

Thank you

Brian

QIDITECH, 2024-01-13:
Ok, we will consider your suggestion if there is enough space on the display.

pmbroth, 2024-01-17:
can be as simple as      Layer 23/450

CChen616, 2024-02-02:
The latest version of fluidd now includes built-in functionality to display the current layer and total layers. For instance, my version of fluidd is as follows:
```
fluidd: v1.19.0-6184c739
```
And the print status looks like

[image]

Regarding support for printer screens, due to the need to upgrade moonraker and considerations for the stability of older models, it may take a long time to test. But this feature will be incorporated into our next-generation machines.

Fisheye3D, 2024-02-02:
Why not release a guide on how to update the mcu on the printhead to the latest klipper so everyone can use updated functionality,  this next generation reasoning makes no sense when it's software and easily updated.

CChen616, 2024-02-03:
The printhead board we currently use is not on the list of boards supported by Klipper for firmware updates via command line. It requires disassembly and connection to another device for updating. If some users opt not to update the firmware while others do, it becomes challenging to maintain update packages for both versions simultaneously.

Fisheye3D, 2024-02-03:
I'm confused, It doesn't need a connection to another device, You can hit the boot button on the device and it mounts to the mkspi as a drive, you can then copy over the klipper.uf2 over to the mounted drive. Then it's updated.

CChen616, 2024-02-03:
Apologies for the confusion, but updating Klipper overwrites many Klipper files, resulting in the loss of our modifications to Klipper. Even after resolving these conflicts, due to some users not wanting to go through the hassle of updating MCU firmware for Klipper, we need to maintain two versions of the update package. Given our current update methods, this will be a significant amount of work.

pmbroth, 2024-02-13:
Ye, I know Fluid has that, but I do not always have the machine on.  Also, so this means we will never be able to upgrade to a new kliiper/moonraker version?  That is really poor design if that is true.

closed 2024-02-19 by CChen616