QIDITECH/QIDI_MAX3 #29: Adding Qidi to Home Assistant

Mike3987, 2024-02-03:
Can you add a Home Assistant integration?

QIDITECH, 2024-02-04:
Sorry, this function is temporarily unavailable.

closed 2024-02-04 by QIDITECH

autoferrit, 2024-03-13:
youcan get this using the "moonraker" plugin from installing HACS. 

HACS: https://hacs.xyz/docs/setup/prerequisites
Moonraker Integration: https://github.com/marcolivierarsenault/moonraker-home-assistant

Install HACS first, then once setup go to its config page, then click the add on store in bottom right, search for "moonraker" and select download. reload HA then add "moonraker" intagrations as any other integration. for the IP it needs the local ip such as "192.168.x.x:10088" and it should just work. @Mike3987