QIDITECH/QIDI_MAX3 #30: Timelapse not working with 4.3.13

VanAustee, 2024-02-04:
Hello, I have the timelapse function configured and working fine till the last update.
I can see the camera and have all the settings in Fluidd, but the printer is not taking any frames anymore, not in the layer timelapse mode nor in hyperlapse.
Can you help me?
Ty

QIDITECH, 2024-02-07:
In this update, we have not changed the timelapse function, please make sure that the printer memory is sufficient, and your operation is correct.
If the above problem does not exist, please try to send instructions through fluidd to shoot, and tell us the screenshot of the result.

VanAustee, 2024-02-08:
I did not change anything in my settings either. 
I cannot shoot via Fluidd as well.
Any idea? Should I try re-installing the timelapse code via SSH?
Below some screens
[image]
[image]

QIDITECH, 2024-02-09:
The plug-in appears to be working properly, possibly because a firmware update caused some configuration errors, please try to reinstall the plug-in.
I'm terribly sorry. We're on holiday. We will fix this issue as soon as possible, after the holidays are over.

VanAustee, 2024-02-09:
I will try to reinstall the plugin. No need to excuse, hopefully it will work again.
If it can help solving the issue, i noticed the new firmware seems to not recognize the two commands:
[image]

VanAustee, 2024-02-12:
Just to understand the steps, you did a factory reset via the touchscreen and then updated the firmware again via USB key. This solved the timelapse issue without reinstalling the timelapse code?
If you confirm i'll try the procedure asap.

> The same thing happened to me. I was only able to solve it by doing a factory reset and updating the firmware.

pganns, 2024-02-13:
I had to reconfigure the klipper on the XMAX 3, as I did when I installed the camera. On the other printer I have, a Qidi Xplus 3, the timelapse continued to work and I didn't need to do a factory reset.

I had opened this issue, which has now been closed by Qidi staff.

https://github.com/QIDITECH/QIDI_MAX3/issues/27

VanAustee, 2024-02-13:
I reinstalled the timelapse plugin and it is working now. Thank you all!

closed 2024-02-19 by CChen616