QIDITECH/QIDI_MAX3 #33: A option for manual leveling.

r00tdownx, 2024-02-13:
Hi do you think its possible for you to put in a button that runs a sequence to level the bed without the top sheet on?

I mean, manual leveling by the screws for the metal plate.  It's really a pain to move the extruder by hand and level the metal plate.

Thanks!

QIDITECH, 2024-02-14:
Ok, we'll take your comments into consideration.

Fisheye3D, 2024-02-14:
> Hi do you think its possible for you to put in a button that runs a sequence to level the bed without the top sheet on?
> 
> I mean, manual leveling by the screws for the metal plate. It's really a pain to move the extruder by hand and level the metal plate.
> 
> Thanks!

You already can, Just add this to your config `[screws_tilt_adjust]
horizontal_move_z: 4
screw_thread: CCW-M4
speed: 300

screw1: 7,37
screw1_name: front left
screw2: 262,37
screw2_name: front right
screw3: 262,297
screw3_name: rear right
screw4: 7,297
screw4_name: rear left`

Then in your console use the command SCREWS_TILT_CALCULATE it will probe all 4 corners and give you a readout to tell you how much to turn the screws.

Anelke, 2024-02-16:
wo füge ich es ein, denn wenn ich es in der printer.cfg einfüge schreibt er mir brauche min 3 schrauben ?

Fisheye3D, 2024-02-16:
> wo füge ich es ein, denn wenn ich es in der printer.cfg einfüge schreibt er mir brauche min 3 schrauben ?

Then you didnt add it correctly because there is 4 screws in the macro. https://www.klipper3d.org/Manual_Level.html#adjusting-bed-leveling-screws

Anelke, 2024-02-26:
es klappt nicht, oder ich bin zu d.....! 
lG

fran6p, 2024-03-04:
@Anelke 
Don't use point but comma in each coordinates:
7.37 => 7,37 …

closed 2024-04-29 by CChen616