QIDITECH/QIDI_MAX3 #35: Saving zoffset reverts back to zero

nutbox76, 2024-03-08:
Hi
I have just received my max 3 and printed of a test piece after setting the z offest via the screen and it printed lovely with no problems then i updated the firmware to the latest version and now after doing the z offset it resets it to zero in the config and i have to manually enter it under the save config line.
Please may you get this sorted because there are many people complaining about this on the facebook groups so is obviously a problem world wide.
Thank you

CChen616, 2024-03-09:
If you are referring to this section under **SAVE_CONFIG**:
```
#*# [probe]
#*# z_offset = 0.000
```
That's because we are now using this section in `/home/mks/klipper_config/config.mksini`:
```
[babystep]
value                          = -1.300
```
Instead of original klipper configuration. The actual z_offset is the sum of these two values.

nutbox76, 2024-03-09:
I'm even so the system isn't saving the offset because when you start the print the nozzle is to high.

CChen616, 2024-03-09:
Could you please perform a printer leveling and then upload the `printer.cfg` and `config.mksini` files as attachments? Due to GitHub's format requirements for attachments, you could rename them as .txt files or add them to a zip archive.

nutbox76, 2024-03-09:
Well I reinstalled the update and then did the bed levelling and it has saved the offset ,for now it is printing well .
If I need to adjust the offset is it best to do it via the max 3 screen or can I do it through fluidd and then save_config?
Also may I ask why qidi has decided to do the z offset like this and not the klipper standard way?
Thank you

trancified, 2024-03-10:
For leveling and z-height, always use the screen and not fluidd.

CChen616, 2024-03-11:
In the standard approach, Klipper requires a restart after leveling to apply the latest Z-offset, which could lead to a loss of connection between the screen and Moonraker, causing the leveling interface to freeze. For users who prefer leveling with Fluidd, it's advised to manually clear the babystep section in `config.mksini` and follow Klipper's official guidelines for leveling. It is generally recommended to continue using the screen for leveling.

nutbox76, 2024-03-11:
So are you saying that you can use the standard way of setting z offset via fluidd and it will save it and use that instead of babysteps?
I ain't bothered about the screen because I would use fluidd to do every anyways.

CChen616, 2024-03-11:
If you opt for leveling using the standard way, please note the following two points:

1. Before starting the leveling process, set the `babystep` value to 0 (if you stick to using the standard way for leveling, you only need to set this once).

2. After leveling is complete, the Fluidd page will display a Z-offset value. Check the `printer.cfg` file; if the value has not been saved correctly, please enter it manually.

For specific leveling instructions, refer to the `Calibrating probe Z offset` section in the Klipper official documentation:
[Probe Calibrate - Klipper documentation](https://www.klipper3d.org/Probe_Calibrate.html)

Waynedejager1, 2024-03-19:
@nutbox76  I have just experienced the exact same issue as you described. 

I tried to follow the chat but it seems confusing , I'd rather revert back to the original firmware I received with the printer ? Can't recall what version it was though? Is it safe to roll back 1 version and then expect z offset to save and work properly?
@CChen616 @QIDITECH @trancified

Waynedejager1, 2024-03-19:
I bought this printer to be free of the creality bullshit I've suffered in the past with never ending tweaks required. 

Was beaming ear to ear when I got it and it worked out the box. The upgrade firmware has killed all that. Wish I stuck to ( if it ain't broke don't fix it ) instead I thought the shiny new toy would be even better. Damn -it. 

Seems like I'll need to follow the retarded workaround.

@nutbox76 has it been an easy fix for you? Is there anyway you could direct me on what to do?

nutbox76, 2024-03-19:
I ended up factory reset and them reinstalled the update and the did the auto bed level and it saved the babysteps  in mks.ini file .
But as said you can 0 that and use klippers z offset through fluidd

Waynedejager1, 2024-03-20:
> I ended up factory reset and them reinstalled the update and the did the auto bed level and it saved the babysteps in mks.ini file . But as said you can 0 that and use klippers z offset through fluidd

The factory reset does nothing on my side the firmware stays at 4.3.13 and the firmware update button doesnt appear to re-upload? any suggestions team - please i want to get printing again

nutbox76, 2024-03-20:
Post up your printer.cfg and mksini and I'm sure someone will get back to you

CChen616, 2024-03-20:
On firmware 4.3.13, factory reset will reset the `config.mksini` file and clean the cached printing history. The `printer.cfg` will not be affected. 

Regarding update detection, please ensure that files such as `QD_Max_SOC` and `QD_Max3_UI5.0` are located within the `QD_Update` folder in the root directory of your USB device. Then, proceed with detection and updating.

Currently, it is recommended to choose and stick to one of the following two methods for leveling your printer:

1. Use the buttons provided on the screen.

2. Follow the method mentioned here using fluidd.

> If you opt for leveling using the standard way, please note the following two points:
>
> 1. Before starting the leveling process, set the `babystep` value to 0 (if you stick to using the standard way [...]

sebdpx, 2024-05-01:
bonjour,
j'ai du reflasher le EMMC et maintenant effectivement le Z offset ne s'enregistre plus .
Doit on le mettre manuellement ici dans le printer.cfg
[image]
pour mon -0.81 

ou dans le baby step ici
[image]
mais mon fichier est très bizarre :
[image]

merci pour votre aide.

CChen616, 2024-05-02:
@sebdpx Please perform a factory reset and then execute the `set z offset` command on the display calibration page. This will ensure that your z offset is saved in `config.mksini`. Alternatively, you can manually enter your z offset in the babystep option of `config.mksini` after reset if you already know the z offset.

sebdpx, 2024-05-05:
Bonjour Merci cela a fonctionné.Le fichier étais sûrement corrompu.Sébastien Dal PraAqua d'OcEnvoyé de mon iPhone