QIDITECH/QIDI_MAX3 #36: Max Volumetric Speed calibration size too big

Mike3987, 2024-03-13:
In the Qidi Slicer, I click calibration under the menu. There is a new option called Max Volumetric Speed. The object does not fit in the x-smart 3 print bed. I cannot do the calibration.

QIDITECH, 2024-03-14:
Yes, this is the standard test model, so we need to test if scaling down the model will affect the test.
We will fix this issue in the next release. Disable or reduce.

Fisheye3D, 2024-03-14:
> Yes, this is the standard test model, so we need to test if scaling down the model will affect the test. We will fix this issue in the next release. Disable or reduce.

Scaling down doesn't effect the test as its a based on layer height not time per layer.

QIDITECH, 2024-03-15:
We have added this feature and will be releasing a new version next week.

closed 2024-03-15 by QIDITECH