QIDITECH/QIDI_MAX3 #38: Install latest Debian + Klipper

ffries, 2024-03-19:
Dear all,

Is there a possibility to install latest Debian with a full stock Klipper upgraded to latest version. I don't mind loosing the screen. Is there a tutorial to build a full new Debian system and install it on a 32G eemc. I don't mind loosing garantee. The Z-Offset way to bed level is too annoying, I want a full Klipper with a traditional bed leveling.

Kind regards,

Fisheye3D, 2024-03-19:
You can upgrade to the latest version, you just need to flash the toolhead mcu, You can also set up a klipperscreen to run through wifi connection to mainsail. Just set the Z offset for screen to 0 and then save it the normal way instead of using there screen use the UI.

CChen616, 2024-03-20:
Upgrading to newer Debian versions, such as Debian 11 or Debian 12, seems to result in issues like network adapter drivers not functioning properly. We are still in the process of testing how to build a stable system with newer Debian version.

Regarding Klipper updates, you can refer to the official Klipper documentation. For the firmware flashing config, you can follow the instructions in this post.
https://github.com/QIDITECH/klipper/issues/1

leadustin, 2024-03-20:
For a current Klipper, Moonraker etc visit this repository. https://github.com/Phil1988/FreeQIDI.git
If you understand German, you will also find instructions here on how to update the system to the latest version, including the touchscreen. [Klipper-Update/update+upgrade.md](url)

Fisheye3D, 2024-03-20:
> For a current Klipper, Moonraker etc visit this repository. https://github.com/Phil1988/FreeQIDI.git If you understand German, you will also find instructions here on how to update the system to the latest version, including the touchscreen. [Klipper-Update/update+upgrade.md](url)

I didn't think the touchscreen worked "The stock display will be out of order until someone reengineers it. "

leadustin, 2024-03-20:
> I didn't think the touchscreen worked "The stock display will be out of order until someone reengineers it. "

Just follow my guide and you'll have a working touchscreen. However, you will need a 5-inch touchscreen and a Raspberry Pi. And very important! Knowledge of German. But with Deepl.com you can translate it.

Fisheye3D, 2024-03-20:
> > > For a current Klipper, Moonraker etc visit this repository. https://github.com/Phil1988/FreeQIDI.git If you understand German, you will also find instructions here on how to update the system to the latest version, including the touchscreen. [https://github.com/leadustin/QIDI-up2date.git](url)
> > 
> > 
> > I didn't think the touchscreen worked "The stock display will be out of order until someone reengineers it. "
> 
> Just follow my guide and you'll have a working touchscreen. However, you will need a 5-inch touchscreen and a Raspberry Pi. And very important! Knowledge of German. But with Deepl.com you can translate it. https://github.com/leadustin/QIDI-up2date.git

Ahh Ok, I see now.. Yeah I have klipperscreen running on a extra rasp pi and touchscreen next to the printer connection over wifi.

Fisheye3D, 2024-03-21:
> Just follow my guide and you'll have a working touchscreen. However, you will need a 5-inch touchscreen and a Raspberry Pi. And very important! Knowledge of German. But with Deepl.com you can [...]

I was looking through your guide, I cant find the printer.cfg you cleaned up and referred to in the guide.

leadustin, 2024-03-22:
Now in english 
https://github.com/leadustin/QIDI-up2date-english

Fisheye3D, 2024-03-23:
> Now in english https://github.com/leadustin/QIDI-up2date-english

Can you tell me how to get the front panel off to remove old screen and replace it? can only reach the bottom two screws on stock screen.

leadustin, 2024-03-23:
Just as you unscrew the two lower screws with an Allen key, unscrew the two upper screws. The front panel does not need to be removed for this. Just feel where the screws are.

Zaros67, 2024-04-12:
no news ... ?

leadustin, 2024-04-12:
> no news ... ?

What kind of news are you talking about?

Zaros67, 2024-04-12:
News firmware update klipper V12 and co

leadustin, 2024-04-13:
> News firmware update klipper V12 and co

I guess not with this generation of printers. After Q1 sure but for series 3 dont think so

jensbodal, 2024-04-18:
Do you mean the X Max 3 won't be getting any additional official firmware updates?

CChen616, 2024-04-18:
> Do you mean the X Max 3 won't be getting any additional official firmware updates?

The upcoming updates we are preparing include support for QIDI Link, power loss resume feature, and adjustments to the UI. However, updates to Klipper or Moonraker are not included in the schedule. The release of these updates may take some time.

As a product that has been on the market for a while, we want to maintain its stability. However, we are considering releasing more comprehensive tutorials (compared to those previously provided) to update Klipper and Moonraker in the future.

leadustin, 2024-04-18:
> Do you mean the X Max 3 won't be getting any additional official firmware updates?

I'm honest and wouldn't get my hopes up too much.

cptcl, 2024-04-18:
> > Do you mean the X Max 3 won't be getting any additional official firmware updates?
> 
> I'm honest and wouldn't get my hopes up too much.

i would kill for an update and maybe an option to get/buy "Dual Sensor Auto Bed Leveling" upgrade for the X-Max 3 ... This would be such a perfect machine

leadustin, 2024-04-19:
> > > Do you mean the X Max 3 won't be getting any additional official firmware updates?
> > 
> > 
> > I'm honest and wouldn't get my hopes up too much.
> 
> i would kill for an update and maybe an option to get/buy "Dual Sensor Auto Bed Leveling" upgrade for the X-Max 3 ... This would be such a perfect machine

For an update, all you have to do is follow my guide. Just check my repository

jensbodal, 2024-04-19:
> > Do you mean the X Max 3 won't be getting any additional official firmware updates?
> 
> I'm honest and wouldn't get my hopes up too much.

Crazy I thought the model was essentially less than a year old after the relaunch

leadustin, 2024-04-19:
> > > Do you mean the X Max 3 won't be getting any additional official firmware updates?
> > 
> > 
> > I'm honest and wouldn't get my hopes up too much.
> 
> Crazy I thought the model was essentially less than a year old after the relaunch

Take a look at creality. They are also churning out new printers all the time. It's probably cheaper to release new printers than to completely update the software on existing devices. The support alone for problems will cost money without end.

billkenney, 2024-05-10:
> Upgrading to newer Debian versions, such as Debian 11 or Debian 12, seems to result in issues like network adapter drivers not functioning properly. We are still in the process of testing how to build a stable system with newer Debian version.

@CChen616 I was able to get the WiFi working on Bookworm like this. I'm not sure whether the make install_fw is necessary. 
1. sudo ln -s /usr/src/linux-headers-6.7.5-edge-rockchip64/arch/arm64 /usr/src/linux-headers-6.7.5-edge-rockchip64/arch/aarch64 ; sudo apt install make gcc build-essential git ; git clone https://github.com/lwfinger/rtl8xxxu ; cd rtl8xxxu ; make clean modules ; sudo make install ; sudo make install_fw
2. Restart the printer, then cd rtl8xxxu ; sudo modprobe rtl8xxxu_git

CChen616, 2024-05-11:
> @CChen616 I was able to get the WiFi working on Bookworm like this. I'm not sure whether the make install_fw is necessary.
>
> 1. sudo ln -s /usr/src/linux-headers-6.7.5-edge-rockchip64/arch/arm64 [...]

I‘ll test this when making a new image for 3 series.

Phil1988, 2025-01-24:
Most of you being already part of the familiy but @ffries @jensbodal and @Zaros67 and everyone else:
Maybe [**FreeDi**](https://github.com/Phil1988/FreeDi) is something for you.

jensbodal, 2025-04-14:
> Most of you being already part of the familiy but [@ffries](https://github.com/ffries) [@jensbodal](https://github.com/jensbodal) and [@Zaros67](https://github.com/Zaros67) and everyone else: Maybe [**FreeDi**](https://github.com/Phil1988/FreeDi) is something for you.

Thanks! I'm finally looking into this now :)