QIDITECH/QIDI_MAX3 #39: Avoiding collisions on Z

ffries, 2024-03-19:
Hello,

I am using latest version of firmware on a QIDI MAX 3.

Several time, the bed left up and did not stop, motors were roaring and I had to stop in urgency. 
After an emergency stop, I did a G28+SCREWS_TILT_CALCULATE and it showed a change in Z.

Kind regards,

CChen616, 2024-03-20:
This could happen in various situations, including incorrect Z-offset setting, PEI plate not installed, damaged mechanical components like Z-axis rods, screws, or motors. 

To better diagnose the problem, could you please provide a detailed description of the circumstances under which the issue occurs? This could help in pinpointing the exact cause and finding an appropriate solution.

potens1, 2024-03-30:
I know it happened to me once, the power cut while first boot assistant was running so the bed was lifted, when I restarted it, it just bumped into the head and did not stop until I unplugged the printer

closed 2024-04-29 by CChen616