QIDITECH/QIDI_MAX3 #44: No Z lift from purge line to the startpoint of the print. If I use the travel lift and lift hight results in a crash

Stoneprophet, 2024-03-28:
Hello,

I have encountered the following problem. I would like to print a label in 2 colors. The first color is printed and then the rest of the object is printed over it. This works perfectly.

The problem is that the print head ignores or ignores the lift height from the end of the purge line to the start point of the print and thus collides with what has already been printed.

I don't think this is intended. The first lift to the specified lift height must take place directly after the purge line. At the moment the print head moves with the distance of the first layer height (0.2mm) to the starting point as far as I can see.

Regards Alex

Stoneprophet, 2024-03-28:
I changed the G-code by myself. Now it works, but you have to know about that bug and other people can't do changes in G-code.
To find out that "lift height" do not work probably, it needs a crash first. Otherwise you will not notice that.

CChen616, 2024-03-29:
It's challenging to provide a detailed analysis without understanding your specific process. 

In QIDISlicer version 1.1.2, when you click on `Preview` and then select a specific height, clicking the `+` symbol on the height line will automatically insert the `Color Change G-code` set in the `Custom G-code` section, which defaults to the M0 pause G-code for easy filament change.
[image]
[image]
From your description, it seems you might be experiencing issues with multi-color printing due to using a different method. Please provide more details about your process, the slicing software, and the version you're using.

Stoneprophet, 2024-03-29:
No it is no multi color printing as intended. My used Technique works till 1mm height without problems. Higher I didn't test.

I made some pictures with explanations. I hope that will help you to understand hot I do that.

Here is a video from the technique. I copy that to the QIDI Slicer. 

[image]

[image]
[image]
[image]
[image]
[image]
[image]

QIDITECH, 2024-03-30:
Thanks for your feedback, we will fix this problem in the next version of the QIDISlicer.

closed 2024-07-25 by QIDITECH

Stoneprophet, 2024-07-25:
I tested it by chance yesterday with the new version and it now works perfectly without any changes. Thank you, that's how I imagine good support.