QIDITECH/QIDI_MAX3 #45: Control board Fan constantly on. Can you disable it when not in use?

4eyez19, 2024-04-08:
There is a fan, i think its for the control board that constantly runs, even when the printer is in sleep/standby mode.

The bambu x1c had the same issue when it first released but they updated it to be silent when not in use.

is there a way to disable this as its very annoying and waste of power.

Thanks

Fisheye3D, 2024-04-08:
> There is a fan, i think its for the control board that constantly runs, even when the printer is in sleep/standby mode.
>
> The bambu x1c had the same issue when it first released but they updated it [...]

Its connected to an always-on non-pwm fan port so no way to disable it.

4eyez19, 2024-04-08:
> Its connected to an always-on non-pwm fan port so no way to disable it.

thats unfortunate :(

I guess there is no other solution?

CChen616, 2024-04-08:
I'm sorry, but yes, currently, the mainboard on the Max model does not have spare pins for fan control, and it uses a direct power connection instead. However, you can contact after-sales service to replace the fan in order to minimize noise caused by dust accumulation or other reasons.

4eyez19, 2024-04-08:
> I'm sorry, but yes, currently, the mainboard on the Max model does not have spare pins for fan control, and it uses a direct power connection instead. However, you can contact after-sales service to replace the fan in order to minimize noise caused by dust accumulation or other reasons.

is the led light powered by the same voltage as the fan? if so, could i join them together somehow so when the light is on when waking the machine, the power will go to the fan also?

CChen616, 2024-04-08:
Yes, they are the same.

leadustin, 2024-04-22:
> Yes, they are the same.

Can you provide a diagram of the X-Max/Plus-3? In the official wiki there is a picture of the mainboard with numbered pins. But there are a few plug contacts that are not described. 2 pieces to the left of connector 16, 1x to the right of connector 16. Then there are 2 USB connectors next to pin 4. I would like to know what the technical specs are.

Fisheye3D, 2024-04-22:
> Can you provide a diagram of the X-Max/Plus-3? In the official wiki there is a picture of the mainboard with numbered pins. But there are a few plug contacts that are not described. 2 pieces to the [...]

Are you talking about this? 
[image]

leadustin, 2024-04-22:
> Are you talking about this? [...]

That's exactly what I meant. Any idea what the H-USB and M-USB ports are for?

Fisheye3D, 2024-04-22:
> That's exactly what I meant. Any idea what the H-USB and M-USB ports are for?

That has been a mystery ive been trying to figure out myself, Nothing on the skpr board these are cloned from gives any clues either.

CChen616, 2024-04-23:
They are used for debugging different chips.
[image]

leadustin, 2024-04-23:
> They are used for debugging different chips. [...]

Is there a way to use PA 11 and PA12 to control a fan/sensor etc.? Is the port voltage 5 volts as standard?

CChen616, 2024-04-23:
As a debugging port, it does not provide sufficient voltage output. You might consider paralleling it with an existing LED light or fan. If you want to install additional sensors, you could consider paralleling several fans together to free up pins for the sensors.

closed 2024-04-29 by CChen616