QIDITECH/QIDI_MAX3 #46: Update klipper , Fluid and aothers as indicated

pmbroth, 2024-04-27:
Hello,  The need to keep the O/S and other related software is needed in order to use the new features of klipper.  

Someone has documeneted how to do it, without losing the diplay functionlaity and using external equipment.

https://github.com/billkenney/update_max3_plus3?fbclid=IwZXh0bgNhZW0CMTAAAR2GaBU-VG_uztrEvHR7IQOvCw-9MTA6IXxXoDkXGHPtS0PxEohWQSAwJoo_aem_AbKazR4Mdu5ryE5o9XUE5CuaDiF2Jj3OX4KT38DlLjSOo_NigIm61h1yaDVxyl6-d-IQ7x7zOpS2hPT5xn7i4OAf

cptcl, 2024-05-14:
Would rly like to see that to be done from official side. I'm thinking hardly to try this guide on my own the next few days. Cause.. If it's that easy.. Then qidi could do an official update in like 2 - 5 hours including testing it.

CChen616, 2024-05-14:
We will use this as a reference to build a new version image, but we also need to release an update package based on the old version，and prepare for the UI update. 

Sometimes starting from scratch is much simpler than gradually adding new features to the old version, especially since we can't be sure which version users are updating from or what modifications they might have made. 

This might take some time as I'm also busy with other projects.

Many thanks to all the community members for their suggestions and solutions. I've learned a lot from them.

cptcl, 2024-05-17:
Since im rly rly happy that i switched fom a Kobra 2 Max (my very first 3D printer where i learnd how to 3D Print, bought in December 2023) to an X-Max 3 (Needed something that can handle ASA, PA, PC and all the CF Filaments and the Kobra limits me so much in learning more.., Xmax3 bought in March 2024).. and i like to optimize things. 
Your X-Max 3 litteraly changed my life. Im doing 3D printing just as a hobby but.. im already addicted

I already done the ratchet / dial / silicone damper mod for the heatbed to my X-Max3 and it works great. 
So, im not afraid of doing or try something (mod or alter things) if,.. at least i have a good knowlege what im doing, why im doing it, how its done right and if there is a chance to revert it back if it doesnt work.

Now... this guide from billkenney here [https://github.com/billkenney/update_max3_plus3] is realy driving me crazy to do it..
Its stuck in my brain every day, even in work. :/
Only thing is, im afraid of doing something wrong or dont understand correctly. 
Everyday i visit the guide, go throu it in my head and tell my self.. 
"U can do this, its easy. Everything is there. Today i will do it." and like 10 min later im like ..
"What if i copy over the wrong file? 100% revertable? What if i fiddle around, only to find the official solution in a week or two after?" 
yeah..  Sry for this short insight :D

Sure.. i know... never change a running system... but as i said,.. i like to optimize and bring things to the latest and best condition.

Sorry if this sounds like i try to push for this.. i dont want to send any signals for stress. 
When i said 2-5 Hours,.. i meant for this machine only. Ofc u have may more projects that needs also attention.
Just wanted to be clear here..
But i would lie if i say "i dont check everyday for firmware and slicer updates" cause i do :D 

Just my thoughts about it..

U (CChen616 and also billkenney) and the whole Qidi team did an amazing job and hopefully in the future too.
Thanks again,.. for this great product and support

CChen616, 2024-05-18:
I can provide a tutorial for creating an image here, to help those who want to try but are afraid of messing things up.

This tutorial assumes you have some related knowledge. You will need an SD card reader, an EMMC adapter, and a functional PC. And this is how the adapter is connected to EMMC
[image]
[image]
1. Log into the Printer System and run the following command:
```
systemctl --no-reload enable armbian-resize-filesystem.service
```
This step is crucial for creating a minimal image. Skipping this step will result in an image size equal to your EMMC storage capacity (e.g., a 32GB EMMC will produce an image just under 32GB).

2. Use virtualization software (e.g., VMware Workstation or VirtualBox) to install a Linux virtual machine (VM). You can use an Ubuntu or Debian system. In this guide, we're using Debian 11.3.0, which can be downloaded from [Debian's archive](https://get.debian.org/images/archive/11.3.0/amd64/iso-cd/).

3. Start the VM and follow the installation prompts. Most options can be left as default, but ensure you select "Yes" when asked to install the GRUB bootloader to the MBR. Selecting "No" will prevent the VM from booting.

4. Open a terminal in the VM and run the following command to install GParted:
```
sudo apt-get install gparted
```
[image]

5. Connect the reader to your VM. In GParted, select the EMMC partition (e.g., /dev/sdc).
[image]

Right-click the second partition (the biggest one) and choose "Resize/Move." If the "Unmount" option is available, unmount it first.
[image]

Adjust the "New Size" slightly above the "Minimum Size." Confirm and apply the changes.
[image]
[image]

In the terminal, check the partition details with:

```
sudo fdisk -l
```
[image]

Note the size, start, and end sectors of sdc2. The end sector will be used in the next step.

Use the dd command to create the image. Replace 13971456 with the end sector + 1:
```
sudo dd if=/dev/sdc of=./imgname.img bs=512 count=13971456 status=progress
```

6. Download the created image to your PC. Use image burning software like BalenaEtcher to write the image to EMMC.
This makes any system changes reversible. If anything goes wrong, you can restore the system using the backup image.

pmbroth, 2024-05-18:
Thank you for the above, will  you still be working on an images and scripts for the  masses to update to the latest software?

CChen616, 2024-05-18:
The answer is a definite yes, but I can not give a exact release date now.

pmbroth, 2024-05-18:
Awesome! - Thank you very much, I updated Fluid, and really looking forward to updated main o/s.  I think once you create a process, that is streamlined and modularized with automated scripts, it will be easier for future upgrades as well.

cptcl, 2024-05-18:
> 1. Log into the Printer System and run the following command:
> 
> 
> ```
> systemctl --no-reload enable armbian-resize-filesystem.service
> ```
> 
> This step is crucial for creating a minimal image. Skipping this step will result the filesystem size equal to your image size (e.g., you will have a 16GB system storage on your 32GB emmc if you have resize the sections to 16 GB in following steps).

Thank u so much for this... Just 1 question..
If i want to have the maximum storage capacity (for my prints) on the emmc (i have an 8gb and an 32gb here and want to use the 32gb one ofc) do i need to adjust SDC2 to 32gb in the followed steps or is SDC2 the "space" for the OS and everything (therefore slightly above the minimum) and the printer allocate it self everything whats left availible for storage for the prints? 
Just wants to be sure here 
Again thank u so much for this help

CChen616, 2024-05-18:
In step five, when resizing, we're actually adjusting the storage space allocated to the operating system. This allows us to create the smallest possible image by not including unused storage space.

Enabling armbian-resize-filesystem.service in the first step ensures that the system will automatically utilize the unallocated storage space on the next boot. If you skip this step, the size of your created image will be the size of your system's storage space when flashed.

If you want the image file to be as small as possible while fully utilizing the EMMC space, just follow this tutorial exactly as outlined.

billkenney, 2024-05-20:
> Only thing is, im afraid of doing something wrong or dont understand correctly. 
> 
> Everyday i visit the guide, go throu it in my head and tell my self.. 
> 
> "U can do this, its easy. Everything is there. Today i will do it." and like 10 min later im like ..

@cptcl it only took me like an hour to get the printer updated and working. It took a bit more time to get things like the WiFi working, and some tinkering here and there, but the guide has been updated and should include everything. Aside from the steps you outlined by @CChen616 to flash the mcus, most of my steps you can copy and paste. 

If you have an extra emmc, I wouldn't worry about messing anything up. I'd say the safest thing to do is to create all of the mcu flashing files with the current Qidi klipper version and save those on a flash drive or something (I think klipper.bin is already in the root folder). Then upgrade. If you're unable to get the system working after upgrading, you already have the files to flash your mcus back to stock.

billkenney, 2024-05-21:
@CChen616 I'm almost done creating an updated image, and I'm wondering if the firmware is any different for the max3 and the plus3, aside from the printer.cfg files? I'm wondering whether it's necessary to create an image for the max3 and the plus3 or if I can just create a single image with instructions to flash the mcus and replace the printer.cfg file for the max3 bltouch/probe or the plus3.

CChen616, 2024-05-21:
Yes, they are the same except for the `printer.cfg` files.

billkenney, 2024-05-25:
@CChen616 there is a working image on my repo along with instructions on updating the printer. @cptcl followed the manual installation steps and confirmed he was able to update with a working screen. https://github.com/billkenney/update_max3_plus3

cptcl, 2024-05-26:
@billkenney @CChen616 
True. Got it to work.. postet every problem on billkenney´s git. Had a few hickups but it worked. I followed his guide (the hard one, not the copy & paste with image) and noticed a few things like ConfigParser to configparser (needs to be done otherwise the script would not run throu since python2 to python3 changes) or the thing with the input shaper. 
Everything mentioned in the issue section. 

Took me like 2 days, cause "ignore any errors" can lead to ignoring all errors. 
The "ignore any errors such as configparser errors" is not a good idea, cause then "sudo dpkg -i mksclient-max3.deb" can not run throu bc it is (or was if altered) not python 3 compartible. (merge.py exactly) Also mentioned in the issiue section. (dont know if it matters but if a a script meant to be run, should run, otherwise it has not to be executed)

In the guide its told to do the *qidi patches* for the display https://github.com/billkenney/update_max3_plus3/blob/main/manual.md on step 12. Files from https://github.com/QIDITECH/QIDI_PLUS3/issues/27#issuecomment-2073932891 and https://github.com/QIDITECH/moonraker/issues/1#issuecomment-1985564638 and than uses 2 files from bilkenney´s git.
Did that and the only thing thats not working is the thumbnails on the printer itself. no matter what u put into slicer settings.

Just my thoughs and what ive noticed, maybe someone has an idea or can explain it to me.

im not that good at coding but...
Why is the moonraker/components/file_manager/metadata.py so much diffrent from my original backup one in terms of thumbnail creation? 
Does the original display now works so differently that we need to use RGB convertion? (ok seems fluidd needs that to show thumbnails and data). 
Why using jpg instead of png and an extra file with _QD at the end? needed? just smaller compression?
Why are those files rotated 90° counterclockwise? (commented that out, now the thumbs are in correct orientation)
Why are thumbnails sometimes handeld as img and sometimes as im ? (original file has all over im and new / patch file has im and img) (does it matter? maybe its only an undefined handler. if so it doesnt matter)
IMHO it cant be a huge thing why the thumbnails dont work.

idk but i think its a moonraker problem after patch, how it handles the thumbnails or the display cannot render/aquire the thumbnail´s.
The thumbnail thing is driving me crazy... will try to dive deeper into.

colin-small, 2024-12-07:
I would also like to be able to upgrade to the latest version of klipper and fluid etc on my X max 3. Is this going to be possible to do anytime soon? I see the guides here but I’m not sure if they apply to the X max 3 or not.