QIDITECH/QIDI_MAX3 #47: Reflashing Toolhead from 0.12 to stock

leadustin, 2024-05-07:
Is it possible to flash a tool head flashed to Klipper 0.12 back to the original Qidi Klipper? The original image is installed. One user claims that it is not possible to flash from 0.12 back to stock

CChen616, 2024-05-07:
As far as I know, this is reversible; at least, I do it frequently

scaramonga, 2024-05-08:
> As far as I know, this is reversible; at least, I do it frequently

Great.  Then a detailed guide of how to do so would be nice.  A simple step my step guide showing how to return an X-PLUS 3 back to the stock QIDI version - 4.2.13.  There is too much conflicting info out there.

My system sits borked ATM, with conflicting attempts at trying to flash print head, one method uses 'Katapult', the other not, which is it?  A method for returning the print head to how Qidi has it would be much appreciated.

MCU' errors galore, and this shows something is not right:
[image]

For EMMC I use the image from QIDI - xindi-20230309
For Mainboard I create flash files from the klipper installed on EMMC and install via SD Card slot on mainboard
For Print head I have no clue what direction to take?

Is this correct?, and what other things am I missing, if any?

CChen616, 2024-05-08:
Try updating using the latest update package then flash the MCU according to the instructions here 
https://github.com/QIDITECH/QIDI_PLUS3/issues/27#issuecomment-2073932891

I have not worked with `Katapult`, so not sure about it.

scaramonga, 2024-05-08:
> Try updating using the latest update package then flash the MCU according to the instructions here [QIDITECH/QIDI_PLUS3#27 (comment)](https://github.com/QIDITECH/QIDI_PLUS3/issues/27#issuecomment-2073932891)
> 
> I have not worked with `Katapult`, so not sure about it.

I did, no difference.

CChen616, 2024-05-08:
According to my colleague, if you choose to update the SOC to version 2.0.6 on the Plus3, there won't be an "updating" interface visible, but it will actually perform the update. If you don't take any action and just wait for the update to complete, you should see a prompt to manually restart. At that point, the update should be finished.

If the method to refresh the toolhead board does not work, then I suggest following the procedure provided by Katapult to reflash the firmware.

scaramonga, 2024-05-08:
Sigh.

Updating without visual reference?, not good.

I don't wish to flash head via Katapult, I just want it back to stock, but it seems this is now impossible once one does it.

It's confusing, to me, anyway.

What are the actual MCU's that get flashed?  Let's start with that, and more importantly, let's list what versions each should be on, stock. (4.2.13) as of today.

EMMC - this is easy, and Qidi have 'xindi-20230309' image for this.  Easily flashed, I use Balena, but Rufus, or most will do.  I also have the official EMMC adapter to do this with.

MAINBOARD - Easy also, flashes fine, manually only for me, as nothing else works.

MCU Extruder - A maze of confusion, with one saying do this, the other do that, and one using 'Katapult', the other not? < -This is what we need clarification on.  Katapult is not 'stock', and in no way returns the head to Qidi defaults.  One guide says to press 'BOOT' & 'RESET', the other just relies on 'BOOT'?  I'm sorry, but unless I'm totally stupid?, the guides conflict, and are not clear, hence confusion.  We need a clear and concise guide regarding the MCU for print head.

MCU RPI - What the hell is this?  Not even mentioned in other guides?, and the one I'm looking at makes no sense?

What are the SOC/MCU/UI figures supposed to look like when 'stock', QIDI, has been successfully flashed to all?

A 'stock' MCU Extruder flash would probably solve this?, but I fear once flashed with anything, then we are where we are, waiting on a replacement board :)

Fluidd as I'm at:

[image]

Fisheye3D, 2024-05-09:
> Sigh.
> 
> Updating without visual reference?, not good.
> 
> I don't wish to flash head via Katapult, I just want it back to stock, but it seems this is now impossible once one does it.
> 
> It's confusing, to me, anyway.
> 
> What are the actual MCU's that get flashed? Let's start with that, and more importantly, let's list what versions each should be on, stock. (4.2.13) as of today.
> 
> **EMMC** - this is easy, and Qidi have 'xindi-20230309' image for this. Easily flashed, I use Balena, but Rufus, or most will do. I also have the official EMMC adapter to do this with.
> 
> **MAINBOARD** - Easy also, flashes fine, manually only for me, as nothing else works.
> 
> **MCU Extruder** - A maze of confusion, with one saying do this, the other do that, and one using 'Katapult', the other not? < -This is what we need clarification on. Katapult is not 'stock', and in no way returns the head to Qidi defaults. One guide says to press 'BOOT' & 'RESET', the other just relies on 'BOOT'? I'm sorry, but unless I'm totally stupid?, the guides conflict, and are not clear, hence confusion. We need a clear and concise guide regarding the MCU for print head.
> 
> **MCU RPI** - What the hell is this? Not even mentioned in other guides?, and the one I'm looking at makes no sense?
> 
> What are the SOC/MCU/UI figures supposed to look like when 'stock' QIDI, has been successfully flashed to all?
> 
> A 'stock' MCU Extruder flash would probably solve this?, but I fear once flashed with anything, then we are where we are, waiting on a replacement board :)
> 
> Fluidd as I'm at:
> 
> <img alt="Annotation 2024-05-08 222628" width="634" src="https://private-user-images.githubusercontent.com/83183034/329066185-ec9631f6-64a5-40ba-b853-568fc9af105c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyMTc5NzksIm5iZiI6MTcxNTIxNzY3OSwicGF0aCI6Ii84MzE4MzAzNC8zMjkwNjYxODUtZWM5NjMxZjYtNjRhNS00MGJhLWI4NTMtNTY4ZmM5YWYxMDVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA5VDAxMjExOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcwNGZjMWI1YjZmMTZjYzYxZDIwY2UyMjllOTk3OWQ1ZGY2ZTk1MTg1ZjZjYjlmMTFlMmMyY2IxNGZhMWE3M2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._1uRyYY5JjtndzXI9wj5q59Qk5_1PSlXhMzrs1C0-kc">

I will start by saying both guides give you a warning about updating, as for the extruder.
Both guides state Boot and reset, I'm not sure where you're not seeing this. 
Flashing the RPI is part of Klipper and is fully documented in the Klipper guide. 
You need to follow the guide for flashing the toolhead but without updating to the latest klipper build. Install Qidi's klipper and build and flash the toolhead as the updating guide shows. It will flash the correct build.

scaramonga, 2024-05-09:
> I will start by saying both guides give you a warning about updating. As for the extruder, both guides state Boot and reset, I'm not sure where you're not seeing this. Flashing the RPI is part of [...]

I will start by saying that I acknowledged this in original thread :)

[image]

Note RED line, no 'RESET' referred to ;)

Fisheye3D, 2024-05-09:
> > I will start by saying both guides give you a warning about updating. As for the extruder, both guides state Boot and reset, I'm not sure where you're not seeing this.
> 
> I will start by saying that I acknowledged this in original thread :)
> 
> <img alt="Annotation 2024-05-09 024756" width="685" src="https://private-user-images.githubusercontent.com/83183034/329099874-25880023-0186-4108-87c3-838758850a7b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyMjAxNjMsIm5iZiI6MTcxNTIxOTg2MywicGF0aCI6Ii84MzE4MzAzNC8zMjkwOTk4NzQtMjU4ODAwMjMtMDE4Ni00MTA4LTg3YzMtODM4NzU4ODUwYTdiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA5VDAxNTc0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVkMGVjM2U1Yjg5OGMwZmQ5OGZmZmNlMjMzM2UzNzMxNTNkOTIxYmMwM2QzN2RkZmU5ZjdmZDViOGNmNTk5NDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.evqPZVI0SlTQ1JFZSBYTCSwYyfgVVxlw9yIKVU_wbK8">
> Note RED line, no 'RESET' referred to ;)
> 
> On the other hand:
> 
> <img alt="Annotation 2024-05-09 025456" width="495" src="https://private-user-images.githubusercontent.com/83183034/329100868-cdfb0adc-dfa3-4f22-8474-fb1c2d0e7559.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyMjAxNjMsIm5iZiI6MTcxNTIxOTg2MywicGF0aCI6Ii84MzE4MzAzNC8zMjkxMDA4NjgtY2RmYjBhZGMtZGZhMy00ZjIyLTg0NzQtZmIxYzJkMGU3NTU5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA5VDAxNTc0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUxMGMxMTAxZGQ5ZmQyYmExM2I0NzAxOTliNWNkNDkwNDBjMjk4NTg5YTMyN2IzOWJlZmI5ZDkzNDNlNTEwMDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ieFEhWjTVSVF21AoOii9_5mEQwe9kCQ7l286Bkau5NE">
> Which is it? ;)

If you had to flash the board the first time using both the boot and reset which one would you think it would be the next time you need to put into dfu mode?

scaramonga, 2024-05-09:
> If you had to flash the board the first time using both the boot and reset which one would you think it would be the next time you need to put into dfu mode?

Well no.  Regardless, I've tried both since, it don't matter either way.  The whole point is a return to 'STOCK', which I now understand is not possible, once one flashes the print head.

CChen616, 2024-05-09:
Only the older versions of the SOC are without an update prompt.

A stock version should look like this:
```
SOC: 4.3.13
MCU: 0.10.0
UI: 4.3.13
```
Have you updated successfully yet?

As for Katapult, I'm not sure how to use it, but you might be able to find some information from its official repository:
https://github.com/Arksine/katapult
Currently, I'm uncertain about the status of your print head board, so I can't offer any specific advice. I will continue to look into this plugin.

MCU here in Fluidd page refers to the mcu on the motherboard. I'm not sure why it still can't connect after flashing; perhaps providing a detailed `klippy.log` could help me understand more.

[image]

scaramonga, 2024-05-09:
> Only the older versions of the SOC are without an update prompt.
> 
> A stock version should look like this:
> 
> ```
> SOC: 4.3.13
> MCU: 0.10.0
> UI: 4.3.13
> ```
> 
> Have you updated successfully yet?

Exactly.  Look at my S.O.C version in above image - 2.0.6???  Something is not right.

CChen616, 2024-05-09:
Yes, version 2.0.6 is the original image version and needs to be updated to the latest version for use. Have you tried pressing the update button and waiting for it to complete the update? There's no interface showing progress during the update, but there will be a prompt once the update is completed. Please ensure that the update file is located in the `QD_Update` folder at the root directory of your USB device and is named `QD_Plus_SOC` without any suffixes like `.bak`.

Fisheye3D, 2024-05-09:
If you have Qidis firmware on the emmc and the board you can do the following,  If you have already flashed the toolhead with katipult when you originally did the update you will need to redownload it as qidi's stock image doesn't contain it 

``` git clone https://github.com/Arksine/katapult```

as you have katapult on already you can move on to flashing the klipper image to it. 

```cd ~/klipper
make menuconfig```

Copy the settings 
![image](https://github.com/QIDITECH/QIDI_MAX3/assets/78997080/100a7111-3a2c-4081-a944-26def8526f14)

Q then Y to save. then use
```make clean
make -j4```

use this command to get your serial
```ls /dev/serial/by-id/*```

next install python 3
```sudo apt install python3-serial```

Next you will flash the toolhead using this command, Make sure you put in your serial id. 
```python3 ~/katapult/scripts/flashtool.py -f ~/klipper/out/klipper.bin -d /dev/serial/by-id/usb-katapult_rp2040_YOUR_ID```

Fisheye3D, 2024-05-09:
If you still get the mcu error you might need to flash the linux mcu, Do that with
```
cd ~/klipper/
make menuconfig
```

then under microcontroller arch choose Linux process then save and exit. then use the following commands
```
sudo service klipper stop
make flash
sudo service klipper start
```
this will flash the rpi mcu

scaramonga, 2024-05-09:
> @scaramonga
> 
> If you have Qidis firmware on the emmc and the board you can do the following, If you have already flashed the toolhead with katipult when you originally did the update you will need to redownload it as qidi's stock image doesn't contain it
> 
> ` git clone https://github.com/Arksine/katapult`
> 
> as you have katapult on already you can move on to flashing the klipper image to it.
> 
> ```
> cd ~/klipper
> make menuconfig
> ```
> 
> Copy the settings ![image](https://private-user-images.githubusercontent.com/78997080/329107839-100a7111-3a2c-4081-a944-26def8526f14.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyMjMzNDcsIm5iZiI6MTcxNTIyMzA0NywicGF0aCI6Ii83ODk5NzA4MC8zMjkxMDc4MzktMTAwYTcxMTEtM2EyYy00MDgxLWE5NDQtMjZkZWY4NTI2ZjE0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA5VDAyNTA0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZjNDI4NDNjYmUwNjI3NDQ3YmE4ZjUyODA4YmFmOGJkMmMzZjFmZTdhZDMyZTUzM2ZlOGQ2YzA1MWQxZDFlZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RmfBlg6AqYznLIJzsODP2-XjR56ZhYGXvG6uWwNb4jg)
> 
> Q then Y to save. then use
> 
> ```
> make clean
> make -j4
> ```
> 
> use this command to get your serial
> 
> ```
> ls /dev/serial/by-id/*
> ```
> 
> next install python 3
> 
> ```
> sudo apt install python3-serial
> ```
> 
> Next you will flash the toolhead using this command, Make sure you put in your serial id.
> 
> ```
> python3 ~/katapult/scripts/flashtool.py -f ~/klipper/out/klipper.bin -d /dev/serial/by-id/usb-katapult_rp2040_YOUR_ID
> ```

I don't get any of they settings with 2040

[image]

Fisheye3D, 2024-05-09:
@CChen616 Can you provide the correct flash chip for the toolboard for @scaramonga

CChen616, 2024-05-09:
Just keep it the same as in the picture
[image]

scaramonga, 2024-05-09:
Yup I did, here we go lol

[image]

[image]

I give up.

CChen616, 2024-05-09:
try sync time first
using
```
# Remove any conflicting services
sudo apt remove -y chrony ntp
# Enable NTP
sudo timedatectl set-ntp true
# Set the timezone
dpkg-reconfigure tzdata 
```

scaramonga, 2024-05-09:
Jesus, why?  I've not set up anything, nor installed nothing?  Image from Qidi?  Now I've got to fiddle with more settings?

Thanks for all your help guys, I really do appreciate it! :), but I'm done with this.

I do accept that it was my choice to deviate from the stock variant, and for that, I regret, but blame no one but myself.  I take FULL responsibility for my actions, and blame not a soul otherwise.

I will say this.  PLEASE keep in mind, before you mod, to have a way back, if all else fails :)

leadustin, 2024-05-09:
Incorrect system time seems to be a recurring problem with Qidi. I hope that this will be solved at some point.
@scaramonga: don't give up, this will probably be the last step to success.

billkenney, 2024-05-10:
@scaramonga you can try updating your SOC firmware by downloading mksclient-plus3.deb from here: https://github.com/billkenney/max3_plus3_recovery. Put it on a flash drive, ssh into your printer, and install it with sudo dpkg -i /home/mks/gcode_files/sda1/mksclient-plus3.deb. I had to comment out the line "systemctl disable gpio-monitor.service" in the DEBIAN/postinst script, because that service did not exist on my printer and it caused the firmware update to fail. Could be the reason it's failing on your printer as well. @CChen616 maybe you could update the firmware postinst script to check if the gpio-monitor.service exists, and only disable the service if it does exist.

scaramonga, 2024-05-10:
Thanks.

EMMC OK, flashed tool head OK, now no connection to printer?

billkenney, 2024-05-11:
Does it boot up and the screen works? Is the firmware version updated? Maybe the IP address changed.

scaramonga, 2024-05-11:
Yup, boots, and screen shows normal layout, but nothing works.

I loaded EMMC tonight with their image, flashed print head using these instructions, done the 'time' thing:
https://openqidi.com/books/upgrades-updates/page/updating-and-flashing-the-the-toolhead-mcu

All went fine, apart from, I no longer can connect, nor SSH into printer?  It's set on a fixed IP, shows up in Router etc.

I'm beginning to wonder if EMMC (the new 32G) one I got is borked?  Problems have stemmed from that.  I have the original 8G one, but I wanna keep that as is, and using it results in the above errors.

I may just try and 'image' the 8G variant, wipe it, then proceed?  What do you think?

billkenney, 2024-05-11:
As long as you got the firmware updated you should be able to connect. I'd try writing the Qidi image to the emmc again (whichever one you want to use), and see if you can connect. Someone also said you can write the image to and boot from a flash drive (https://github.com/QIDITECH/QIDI_PLUS3/issues/27#issuecomment-2081588087), and it takes priority over the emmc, so you could try that as well if you dont want to overwrite your emmc.

scaramonga, 2024-05-11:
OK.

Put original EMMC in, connection fine, but:
[image]

Note, both restart buttons, none work.  Still, I have access to system now.

billkenney, 2024-05-11:
My best guess is that you didn't flash the extruder with the 0.10 klipper file. Restart holding the bottom left button for like 1 min. ssh into your printer, cd /home/mks/klipper ; make clean ; make menuconfig. select rpi2040, w25q080 with clkdiv 2, communication interface usb. then make. then ls /home/mks/gcode_files/*. if you see sda1 in the list, cp /home/mks/klipper/out/klipper.uf2 /home/mks/gcode_files/sda1. if you dont see sda1, sudo mount /dev/sda1 /dev/mnt ; cp /home/mks/klipper/out/klipper.uf2 /dev/mnt. then restart the printer

scaramonga, 2024-05-11:
AT LAST!!!! WooHoo, LOL!!!

I tried one more shot at flashing mainboard, and it worked!  Light actually came on in chamber (long time since I've seen that lol).

All seems OK, not printed yet, but no errors thus far.  I would like to know (I'm like that) as to what the problems was?, but I suspect flashing the print head was it.  Fluidd ain't working, neither is connection in Orca, but Rome wasn't built in a day lmao :D

billkenney, 2024-05-11:
If fluidd isn't working, try both port 80 and 10088, i.e., 192.168.1.xxx:80 and 192.168.1.xxx:10088. 

If it still isn't working, ssh into your printer, run the following commands: `cd /home/mks ; rm -rf fluidd ; wget 'https://docs.google.com/uc?export=download&id=1Essar3y789UNelY1G5T_Dro7V41O4m4K' -O fluidd-1.28.tgz ; tar -xzf fluidd-1.28.tgz ; mv dist fluidd ; rm fluidd-1.28.tgz`
Turn your printer off and on again

scaramonga, 2024-05-11:
Good stuff!!  Fluid working dandy ho :)

Hey bud, many thanks for all your help, it is honestly much appreciated!  Superb, and I can't thank you enough :)

billkenney, 2024-05-11:
> What I will say, is, if your gonna mess with this 'mess' of a system, then have that original system backed up!, as I can say, for certain, I'd be F'''d. ;)  Weird way Qidi have done this.

You're welcome. I definitely agree you need a working backup before trying to update to unsupported software. I'm running the latest software with a working stock screen, but I'm sure there are issues other people could run into that I didn't run into.

scaramonga, 2024-05-12:
A little update.

Cloned EMMC image to 32G variant, extended usable space, all good still :)