QIDITECH/QIDI_MAX3 #48: Wiki to clone klipper/moonraker repos after updating

billkenney, 2024-05-13:
There are a lot of people who unwittingly update klipper or moonraker in fluidd not knowing their printer will no longer work. The majority of people seem to think the update has rendered their emmc unusable and they have to purchase a new one. But the fix is simple--delete the klipper/moonraker folder and clone the repo from Qidi's GitHub page. If you get an ssl certificate issue you have to fix the date. You should really add instructions on how to do this to your wiki.

billkenney, 2024-05-16:
@CChen616 here's a simple script that will fix inadvertent klipper/moonraker  updates, and install the latest working version of fluidd.

CChen616, 2024-05-16:
Thanks for your help, but actually, the Klipper and Moonraker repositories on Git are outdated. Additionally, if Fluidd is updated to version 1.28, it is not fully compatible with the current Moonraker (it will work, but there will be a prompt to update Moonraker). We will upload the adjusted components to Git and update the wiki accordingly.

billkenney, 2024-05-16:
The 1.28 version linked to in my script is the last one that works with the moonraker on your GitHub repo, the next commit breaks the webcam. I wasn't aware of an issue with an update prompt. 

I'm glad you'll be updating the wiki. I see people post about this on Facebook at least every 2-3 days, so it will definitely help.

billkenney, 2024-05-25:
You also have to change the buster backports source to avoid apt errors:
sudo sed -i 's/deb.debian.org\/debian buster-backports/archive.debian.org\/debian buster-backports/g' /etc/apt/sources.list