QIDITECH/QIDI_MAX3 #50: Fill Gaps on First Layer

r00tdownx, 2024-05-23:
Hi can you change the slicer to put fill gaps layer after perimeter and infill on first layer or else it really screws up.

Thanks!

[image]
[image]
[image]

QIDITECH, 2024-05-23:
Okay, we'll figure out how to optimize that. If available, please provide the configuration of this file.

r00tdownx, 2024-05-23:
Hi I already change the model and didn't save it... but the gap fill issues happens when you have small objects that don't quite fill using 2 or 1 pass.

This is also a issue here: 
For some reason it starts with all the small parts before doing the larger sections to join it to...  Should always do Islands last on first layer...
[test.zip](https://github.com/QIDITECH/QIDI_MAX3/files/15417859/test.zip)

[image]
[image]

r00tdownx, 2024-05-24:
Hi got the issue again on new file.
[Handle 005_1.zip](https://github.com/QIDITECH/QIDI_MAX3/files/15427484/Handle.005_1.zip)
[image]

QIDITECH, 2024-07-25:
I am very sorry for not replying to you in time due to misunderstanding in communication.
Gap fill will print first because you have enabled Infill before perimeters.