QIDITECH/QIDI_MAX3 #51: Z axis Home at X0 Y0?

r00tdownx, 2024-05-23:
Hi is there a command where I can home the Zaxis at X0 Y0 with BL touch?

I want to resume prints by editing gcode but I cannot home the Z axis as it wants to home in the middle of the plate.

If I can home Z axis in corner maybe I could do a resume on power out but editing gcode?

CChen616, 2024-05-24:
It is not recommended to home the Z-axis during power recovery as it can easily cause crashes. Instead, you can use this plugin. It records the current height at the beginning of each layer and resumes printing from that layer after a power recovery.

https://github.com/The--Captain/plr-klipper

We will include our custom power recovery feature in the next update.

r00tdownx, 2024-05-24:
Ok great!