QIDITECH/QIDI_MAX3 #53: Update Klipper to 0.12 and Python to Python 3

boydenus, 2024-05-29:
The current version of Qidi's Klipper firmware is three years old and needs to be updated to a current version to enable users of Qidi hardware to have access to a multitude of recent bug fixes and feature enhancements provided by the main line software branch. Further, Qidi's firmware uses an out-of-support version of Python 2.7 that has many Critical and High-level security vulnerabilities. Python should be updated to Python 3 to correct these security vulnerabilities and enable use of the main line Python code base and modules.

CChen616, 2024-05-30:
I am currently setting up a new system based on Debian 12 for the MAX3 Plus3 series machines. The system is up and running, and I am currently testing print quality and other aspects. The new image is expected to be available in a few days.

pmbroth, 2024-05-31:
Will you be updating Fluid to the lates as well?

cptcl, 2024-05-31:
And Moonraker? (just to complete the list) :D

CChen616, 2024-06-01:
Currently, this system is based on Debian 12 and includes Python 3.11.2. It has the latest versions of Klipper, Moonraker, and Fluidd installed, along with plugins such as Adaptive Mesh, Crownset, and Spoolman. Is there anything else that needs to be added?

boydenus, 2024-06-01:
I'll let others chime in, but I believe that covers the full stack. Thank you very much for your efforts!

billkenney, 2024-06-01:
> Currently, this system is based on Debian 12 and includes Python 3.11.2. It has the latest versions of Klipper, Moonraker, and Fluidd installed, along with plugins such as Adaptive Mesh, Crownset, and Spoolman. Is there anything else that needs to be added?

moonraker-timelapse, klippain-shaketune, and I haven't installed plr-klipper yet but it would definitely be useful: https://github.com/The--Captain/plr-klipper

leckert123, 2024-06-01:
Auto Z offset to eliminate using the leveling sheets, like Q1P, would be awesome if technically possible.

pmbroth, 2024-06-04:
@CChen616   Hello - wanted to know how testing is going?  Did you look at the above requested items as well?  Thank you for all of the work you are doing!!

Fisheye3D, 2024-06-04:
> Auto Z offset to eliminate using the leveling sheets, like Q1P, would be awesome (if technically possible).
> 
> ![Screenshot 2024-06-01 at 16-09-40 Qidi Tech Q1 Pro 3D Printer](https://private-user-images.githubusercontent.com/41870928/335825381-3736ff06-9827-46fd-9ca7-393dd78424bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1MTUyMTYsIm5iZiI6MTcxNzUxNDkxNiwicGF0aCI6Ii80MTg3MDkyOC8zMzU4MjUzODEtMzczNmZmMDYtOTgyNy00NmZkLTljYTctMzkzZGQ3ODQyNGJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA0VDE1MjgzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZkNGE1MjUwNTdlYTdiODY1NjgxYjRlNWFlN2I0MTA4Njc2NTFlZTdjYTBlMzYwN2E0N2M2ZTk1MjdjY2E4MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.S5fp-H2ECy0heOXZOQRfpIqn6jVFjGrZKRN4aJl1d4U)

If you want auto bed leveling like the Q1 your not going to get it with the stock setup or even updating klipper, You can get a beacon probe and use it for auto z offset

cptcl, 2024-06-04:
> > Auto Z offset to eliminate using the leveling sheets, like Q1P, would be awesome (if technically possible).
> > ![Screenshot 2024-06-01 at 16-09-40 Qidi Tech Q1 Pro 3D Printer](https://private-user-images.githubusercontent.com/41870928/335825381-3736ff06-9827-46fd-9ca7-393dd78424bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1MTUyMTYsIm5iZiI6MTcxNzUxNDkxNiwicGF0aCI6Ii80MTg3MDkyOC8zMzU4MjUzODEtMzczNmZmMDYtOTgyNy00NmZkLTljYTctMzkzZGQ3ODQyNGJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA0VDE1MjgzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZkNGE1MjUwNTdlYTdiODY1NjgxYjRlNWFlN2I0MTA4Njc2NTFlZTdjYTBlMzYwN2E0N2M2ZTk1MjdjY2E4MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.S5fp-H2ECy0heOXZOQRfpIqn6jVFjGrZKRN4aJl1d4U)
> 
> If you want auto z offset like the Q1 your not going to get it with the stock setup or even updating klipper, You can get a beacon probe and use it for auto z offset

Like a 2nd probe or what do u mean?

trboyden-aus2inc, 2024-06-04:
@Fisheye3D and @cptcl can you guys take that conversation offline or open another issue? That's off-topic for this issue request.

Thanks.

Fisheye3D, 2024-06-04:
> @Fisheye3D and @cptcl can you guys take that conversation offline or open another issue? That's off-topic for this issue request.
> 
> Thanks.

Actually it was on topic, someone asked about it being asked after a list was requested by @CChen616 and a response was given to why a request cant be done so please explain why its off topic...

trboyden-aus2inc, 2024-06-04:
@Fisheye3D because this is my issue request and you are hijacking it.

cptcl, 2024-06-04:
> @Fisheye3D because this is my issue request and you are hijacking it.

There are literally 3 other issues open with practical same request. So your issue could be seem as spam than. Just stop it here

Fisheye3D, 2024-06-04:
> @Fisheye3D because this is my issue request and you are hijacking it.

No one highjacked shit, Again someone asked for a feature that was requested by one of the mods to begin with.. A answer was given to why the request cant be done and a solution for it, It was in line with the post to begin with, if anything your comments are off topic and baseless...

Fisheye3D, 2024-06-04:
> Like a 2nd probe or what do u mean?

You can replace your probe with a beacon probe.

cptcl, 2024-06-04:
@Fisheye3D May u have some reference for doing so? Maybe I try doing this if u have. Thx anyway.

leckert123, 2024-06-04:
CChen has been nothing but gracious and polite. And I appreciate the openness. 

However, the likelihood that CChen is going to respond to any additional needs or wants for the firmware update, now that this has completely gone off the rails, is slim to none. Thanks guys.  You've likely chased away a well intentioned developer.

Apologies to CChen. 

I was merely hopeful that perhaps with the rest of the software upgrades, maybe even with with a Qidi-offered inductive probe option, it might be possible to do a Q1P sort of Z offset upgrade. You know, something easy and factory supported.

CChen616, 2024-06-05:
Currently debugging the power loss recovery feature and automating some configurations.

For Q1 model, we used additional sensors to assist in automatically setting the Z offset, which may not be applicable to the Max3 model.

Furthermore, because the Max3 control system stores the Z offset externally (in config.mksini) instead of in the Klipper configuration file (likely to avoid the need to restart Klipper), these solutions may not be applicable :(

leckert123, 2024-06-05:
CChen, thank you for looking in to this for us. Not everything is possible. Maybe one day.... :)

Excited about the power loss recovery. I live in an area with frequent, usually brief, power outages due to strong weather and hurricanes.  It will be a really useful feature for me.

billkenney, 2024-06-05:
@CChen616 someone has asked me if the Smart3 can follow the same procedure to update to the latest software. Are the menuconfig settings the same for the Max3/Plus3/Smart3? If so, as long as they reinstalled the firmware, they should be able to use the same image with a working screen?

billkenney, 2024-06-05:
> For Q1 model, we used additional sensors to assist in automatically setting the Z offset, which may not be applicable to the Max3 model.

If there is one or more sensors we can install on the Max3/Plus3/Smart3 to automatically calculate Z Offset, I'd definitely buy it. That would be an amazing upgrade.

michaelbeals, 2024-06-06:
@billkenney imo they should contact customer support. This isn't the place for that sort of help.

CChen616, 2024-06-06:
> @CChen616 someone has asked me if the Smart3 can follow the same procedure to update to the latest software. Are the menuconfig settings the same for the Max3/Plus3/Smart3? If so, as long as they reinstalled the firmware, they should be able to use the same image with a working screen?
> 
> Does it matter that the motherboard is X-6 vs X-4?

They are basically the same except for the `printer.cfg` and the controlling system `xindi`. They share the same menuconfig. But it's better to keep a backup image before reinstall.

CChen616, 2024-06-06:
Regarding power loss recovery, I spent several days testing and found that due to hardware limitations, the Z-axis of the MAX3 model cannot maintain its height during a power loss, making recovery almost impossible :(

Any solution to this issue?

billkenney, 2024-06-06:
> Regarding power loss recovery, I spent several days testing and found that due to hardware limitations, the Z-axis of the MAX3 model cannot maintain its height during a power loss, making recovery almost impossible :(

@CChen616 I'm waiting on some parts to get my printer back up and running, but I can do some testing in the next 2-3 weeks and let you know if have any success. Were you testing with settings similar to the Q1 or different settings? If you could share those settings, it would give me (and others) a good place to start.

cattivik66, 2024-06-06:
Imho the recovery is not essential, anyway yes the Z motors can’t hold the position without energy. Possibly a small battery to just power the motors in case of power loss.

Honestly I am much more happy for the OS and software updates (in my case for the x-plus 3).

if you want, as small improvement, enable “object exclusion” by default (this can already be done by upgrading fluid to a never compatible release).

CChen616, 2024-06-07:
Here is the new image link:
https://github.com/CChen616/QIDI_Max3_Bookworm

billkenney, 2024-06-07:
> Here is the new image link:
> 
> https://github.com/CChen616/QIDI_Max3_Bookworm

Nice work!

Does the WiFi and gcode preview still work on screen? Those are the two issues I've noticed That do not work with my upgrade image (I have not installed the patch files).

Also, I would assume the Plus3 can also use this image as long as they update the printer.cfg and flash the mcus? And the Smart3 can use this image as long as they update the firmware, update the printer.cfg, and flash the mcus?

CChen616, 2024-06-07:
For the Plus3 and Smart3 models, this image might not be suitable at the moment. It's not just the printer.cfg that needs modification; the main issue is the screen control system. Although I haven't looked deeply into it, there are some differences. Some file paths are hard-coded and need to be changed, and separate UI update files need to be prepared.

billkenney, 2024-06-07:
> For the Plus3 and Smart3 models, this image might not be suitable at the moment. It's not just the printer.cfg that needs modification; the main issue is the screen control system. Although I [...]

I guess we'll have to wait and see if someone with the plus3 or smart3 tries to update and reports back. I'll let you know if I hear anything.

cptcl, 2024-06-07:
@CChen616 The QD_Max3_UI5.0 rename to 800_480.tft and insert in root right?

CChen616, 2024-06-07:
Either renaming and put under /root and restart or put directly into USB and update would work

billkenney, 2024-06-07:
@CChen616 I installed the image, logged in via ssh, tried to install something with apt, and I get a bunch of errors. I thought it was because the sources.list was china, but I changed the sources to default sources, and I still get these errors:

W: Failed to fetch https://apt.debian.org/debian/dists/bookworm/InRelease  Could not connect to 127.0.0.1:7890 (127.0.0.1). - connect (111: Connection refused)
W: Failed to fetch https://apt.debian.org/debian/dists/bookworm-updates/InRelease  Unable to connect to 127.0.0.1:7890:
W: Failed to fetch https://apt.debian.org/debian/dists/bookworm-proposed-updates/InRelease  Unable to connect to 127.0.0.1:7890:
W: Failed to fetch https://apt.debian.org/debian/dists/bookworm-backports/InRelease  Unable to connect to 127.0.0.1:7890:
W: Failed to fetch https://security.debian.org/debian-security/dists/bookworm-security/InRelease  Could not connect to 127.0.0.1:7890 (127.0.0.1). - connect (111: Connection refused)
W: Failed to fetch http://apt.armbian.com/dists/bookworm/InRelease  Could not connect to 127.0.0.1:7890 (127.0.0.1). - connect (111: Connection refused)

CChen616, 2024-06-07:
I used clash and forget to remove these settings. 
```bash
sudo nano /etc/environment
# remove these setting
export http_proxy="http://127.0.0.1:7890"
export https_proxy="http://127.0.0.1:7890"
export no_proxy="localhost, 127.0.0.1, *edu.cn"

sudo visudo
# remove these setting
Defaults env_keep+="http_proxy https_proxy no_proxy"

nano ~/.gitconfig
# remove these setting
[http]
  proxy=http://127.0.0.1:7890
[https]
  proxy=http://127.0.0.1:7890
```

billkenney, 2024-06-07:
@CChen616 Thank you for the explanation. I removed the env_keep+ from visudo. The only thing in env for mks was http_proxy--the other two were not there. I removed this with `unset http_proxy`. Neither of the urls was in ~/.gitconfig. I'm now able to update and install via apt with the mks user

When I `sudo su` and try `apt update` (as root user), I get the same errors. `env` shows the following
no_proxy=localhost, 127.0.0.1, *edu.cn
https_proxy=http://127.0.0.1:7890
http_proxy=http://127.0.0.1:7890

The http and https proxy were not in /mks/.gitconfig, but they were in /root/.gitconfig file, and I removed those lines

After running `unset no_proxy ; unset http_proxy ; unset https_proxy` (as root user) everything updates like it should and I'm able to install programs

dewi-ny-je, 2024-06-07:
> Here is the new image link: https://github.com/CChen616/QIDI_Max3_Bookworm

I read there:

`After careful consideration, our team has decided that system upgrades will not be covered under the after-sales warranty. Therefore, this image is released by me personally.`

I think QIDI took the right decision: all the current printers work pretty well already and all these kind of requests like newest klipper, moonraker, ... are coming from a minority of advanced users which should not result in requests for replacement part of support for modifying the printers.

More important would be to update ALL printers to the latest Debian, even if the "printing stack" (klipper & co) stayed the same, because security should be a MUST in every company in 2024.
That's something I would expect even without any feature change.

boydenus, 2024-06-07:
Thanks @CChen616 really appreciate your work on this! I have a current project running on the X-Max 3, but once that is complete, I will test this out myself.

leckert123, 2024-06-07:
CCHen, I am not clear on the discussion of updates and warranty.

Is the situation #1, #2 or #3 below (or something else entirely):
1. For the X-Max-3 to remain under warranty, Qidi will not issue future firmware updates for a customer to install. The machine must forever stay on the original image shipped with the printer from the factory,  
2. For the X-Max-3 to remain under warranty, a customer must not install any firmware updates that are not officially released by Qidi, including 3rd party, experimental, and the not-released development firmware that you posted?
3. For the X-Max-3 to remain under warranty, there can be no future Klipper / Python firmware updates, or added features.

I worry that if the answer is #1, that X-Max-3 devices are left with no ability to perform firmware updates at all, which seems quite bad.  If the answer is #2 - I completely understand and agree.  If the answer is #3, the machine will be forever stuck on old versions of Klipper and Python, and no new features will be added.

cptcl, 2024-06-07:
> CCHen, I am not clear on the discussion of updates and warranty.
> 
> Which of these situations (or maybe something else) is the concern about updates and warranty?
> 
>     1. For the X-Max-3 to remain under warranty, Qidi will not issue future firmware updates for a customer to install. The machine must forever stay on the original image shipped with the printer from the factory,
> 
>     2. For the X-Max-3 to remain under warranty, a customer must not install any firmware updates unless they are officially released by Qidi.  So no 3rd party, experimental, or even the development firmware that you posted.
> 
>     3. For the X-Max-3 to remain under warranty, there will be some Qidi-released firmware updates to fix bugs, but no future Klipper / Python firmware updates, or new added features.
> 
> 
>     * If the answer is 1, X-Max-3 devices are left with no ability to perform firmware updates at all, which seems quite bad.
> 
>     * If the answer is 2 - I completely understand and agree.
> 
>     * If the answer is 3, the machine will be forever stuck on old versions of Klipper and Python, and no new features will be added, but some bugs may be addressed.  This isn't wonderful, either, but less bad than answer 1.

Uhm...
1. True and it does not shift,, so... keep your hands calm+
2. True.. aigain.. keelp your hands calm if u already have  MAX_V4.3.13.rar  installed
3. True except... "Some bugs may be adressesd" <- this not gona happen....

Yes, Yes, Yes, and Yesnt.... U install an experimanetal version from an "unknown" source (Who try to help the community and try keep this awesome poduct  (witch could be easzly threadet as an OTG update from anywhere.... or maybe QIDI is only a 6 man grop and cant handle a ..(very very good!!!) constructive answer at all.. idk.

TLDR... if u install this.. ur not getting under warranty!
In fact.... if ur trying to update anything onm your own... your not gettin g under warrenty. not a single cent... nothing. (PS: user of an Max 3  since March 2024, with an 3y old OS, Kipper 10 instead of 12 and even maybe this year 13 / 12.5. 

A hero named @CChen616 uses its time, to get u all to a an Up2date version... idk but at least 3 days of trial and error where involved. Only 5 ppl thanked him... No one try to force qidi to an official update...
Sure if u do... yes.,.. ply cancel the  point where u can juist order free replacement stuff from your store... but at  least make it official... 1 in like 3 years...

I have my 8gb ... non touched EmmU so... if somthing break.,...  I never did any updates.... rigth? RIGHT?

Just ---.... PLS QIDI go in the same right direction like Anycubic goes. (or want to go youse there are stupid) .. but just do it faster! U wanna sell? u need to  be... at least compareable.

leckert123, 2024-06-07:
I sort of doubt that there will be no firmware updates ever. That's more than a little bit pessimistic.  There is an entire methodology, based on using the memory stick, to do firmware OS updates.  Aurora Tech showed doing this and she noted how very long the updates took because it replaced the entire OS.  So, it's a thing. 

I guess the real question will be in the future, what sorts of official updates will be released by Qidi and will their own releases invalidate the warranty.

CChen616, 2024-06-08:
From a developer's perspective, I define firmware as a read-only program that can only be modified by directly re-flashing it, such as writing Klipper.bin to the motherboard's MCU.

My daily development work mainly focuses on software running on the operating system, such as modifying Klipper, Moonraker, writing Python and sh scripts, and developing new screen control systems.

Since the bootloader for the Max3 series extruder board and motherboard either doesn't exist or doesn't work well, MCU flashing is currently done manually. Even with the most detailed tutorials, you can't ensure that all users can handle the issues and successfully flash the firmware. This also means providing warranty for potential issues, which is why Klipper updates have not been released. However, this should be improved in the latest generation of machines, which will automate this process.

Therefore, for the Max3 series machines, including Plus3 and Smart3, there likely won't be any future updates for Klipper (which requires firmware flashing), the operating system, or Python (which is hard to update with the existing package method).

Even if this image is perfected and re-released, the cost of providing connection tools and warranty services for potential issues makes it not worth considering.

So, while there may be software updates in the future, there will likely be no firmware updates.

However, for those users who have some understanding of the Linux system and want to update Klipper, Python, the operating system by themselves, etc., I will provide as much help as possible (provided they can handle most of the issues that arise during the process).

leckert123, 2024-06-08:
Thank you for the explanation. 

Would it be possible for new firmware to be supplied on the EMMC so that when the machine boots up it flashes the new firmware correctly?

CChen616, 2024-06-08:
> Thank you for the explanation.
> 
> Would it be possible for new firmware to be supplied on the EMMC so that when the machine boots up it flashes the new firmware correctly?

That would likely involve a motherboard and extruder board with new firmware. If these were to be sold, they would need to undergo thorough testing. Given the available manpower, I believe the company might prefer to invest in the development of new machines (in fact, I might have already spent too much time on the Max3). 😢

dewi-ny-je, 2024-06-08:
If the printer works, it works. 
It's a pity about not having Debian updates but as it concerns klipper, the machine it's sold with certain features which are there, and no enhancement was promised, so nothing is to be expected.

boydenus, 2024-06-08:
> From a developer's perspective, I define firmware as a read-only program that can only be modified by directly re-flashing it, such as writing Klipper.bin to the motherboard's MCU.
>
> My daily [...]

OK, I think that is the clarity a lot of us have been waiting on an explanation for. Basically, it's a custom toolhead implementation so we're screwed, just like with the screen. Unfortunately, what is done is done.  I kind of take issue with using the term open-source around it, because other than the fact it is uses open-source software, everything else about it goes against the open-source methodology. Proprietary firmware and hardware that can't be updated is a big no-no. I suggest Qidi stop using that in their marketing material.

I honestly don't know why all these 3D printer companies are putting any electronics in the tool head. All you need is a bunch of screw terminals or lever nut connectors on a backer plate to make all the relevant connections, or better yet use a quick-change tool head like the BIQU Hermit Crab. If it's for the input shaping, that can be done on the probe now, with lots of options out there including the BTT Eddy, Cartographer, or Beacon probes. 

You all are making it harder on yourselves using proprietary designs versus those that are already compatible with existing Voron designs that work fine with the open-source software and don't lock you into specific versions of software. I'm not saying Qidi should just copy Voron by any means, but certainly use already engineered and proven aspects such as the tool head system and linear rails. They also use a fixed bed design precisely for the reason your power-loss recovery doesn't work - bed is too heavy and loses z-axis. But if a tool head gantry design loses z-axis, no big deal, because it just goes back to the height specified in the last g-code sequence. Vorons are certainly not perfect, but that's the great part of an open-source design - there is an entire Github repo out there where users have already solved those issues for you and contributed all the designs back to the community that anyone can use and build upon.

boydenus, 2024-06-08:
> If the printer works, it works. It's a pity about not having Debian updates but as it concerns klipper, the machine it's sold with certain features which are there, and no enhancement was promised, so nothing is to be expected.

Yeah, that is a bunch of horse shit. I accept the proprietary hardware explanation even if I don't like it. But that's an abuse of open-source software and is greatly frowned upon to the point companies have been sued and banned for doing stuff like that, especially when they call their machines open-source in their marketing material.

boydenus, 2024-06-08:
> > If the printer works, it works. It's a pity about not having Debian updates but as it concerns klipper, the machine it's sold with certain features which are there, and no enhancement was promised, so nothing is to be expected.
> 
> Yeah, that is a bunch of horse shit. I accept the proprietary hardware explanation even if I don't like it. But that's an abuse of open-source software and is greatly frowned upon to the point companies have been sued and banned for doing stuff like that, especially when they call their machines open-source in their marketing material. Using the term open-source in relation to your product comes with expectation that the code is open to inspection, modification, and is updateable - the core tenants of open-source software and written directly in the software license agreement that you automatically accept by using it.

leckert123, 2024-06-08:
From what I understand, every 3D printer company takes Klipper, Moonraker, etc and then modifies it a bit to suit their hardwarem, just like Qidi does.  Qidi states what they do right here: https://github.com/QIDITECH/klipper  

If someone is able to successfully update to the latest and greatest, and comes up with a step-by-step set of flashing instructions, then great.  I might try it (after I'm out of the warranty period).  

Dumb question - I guess I don't know why Klipper or Moonraker, which are software, and presumably reside on the EMMC, require some update to the MCU firmware if the hardware on the machine does not change.

cptcl, 2024-06-08:
@leckert123 i did the update right away and it works. My printer is also not "out of warranty" rn.. but i did quite a few modifications to my printer. so i dont have it anyway. Sure i would rather have an official update, but if this is the only thing we get, ill take it.

Dumb answer - i dont know exactly why but.. my 50 cents.. if klipper / Moonraker changes a way how it communicates with the board or printhead, than sometimes u have to adjust the firmware on that so the comunication establishes, or give the right callbacks/commands.  
On an official update this would run in the background and u would not notice. U would only see the display update, that takes forever :D 

And ur right. The printer works out of the box and the update does not bring that much of a difference to the table. 
Also im not worried about "security" of the OS cause if u think about it,.. nearly every klipper printer has "mks" and "makerbase" for login and so this is already a good attack point for everyone.

I think the same way like i think about pc mainboard bios updates. Pc runs anyway but maybe a bit more stable or bit more performance with newer bios.

boydenus, 2024-06-08:
> From what I understand, every 3D printer company takes Klipper, Moonraker, etc and then modifies it a bit to suit their hardware, just like Qidi does. Qidi states what they do right here: https://github.com/QIDITECH/klipper That's no different than open source Android on phones,.
> 
> If someone is able to successfully update to the latest and greatest, and comes up with a step-by-step set of flashing instructions, then great. I might try it (after I'm out of the warranty period).
> 
> Dumb question - I guess I don't understand why Klipper or Moonraker, which are software that presumably reside on the EMMC, require an update to the MCU firmware if the hardware on the machine does not change.

So, this is where we get into proprietary design of the Qidi. As an example, here is the typical file system from an X-Max 3:

Filesystem     Type     1K-blocks    Used Available Use% Mounted on
udev           devtmpfs    426036       0    426036   0% /dev
tmpfs          tmpfs        99980   10500     89480  11% /run
/dev/mmcblk1p2 ext4       6868908 5192184   1582704  77% /
tmpfs          tmpfs       499892       0    499892   0% /dev/shm
tmpfs          tmpfs         5120       4      5116   1% /run/lock
tmpfs          tmpfs       499892       0    499892   0% /sys/fs/cgroup
tmpfs          tmpfs       499892       0    499892   0% /tmp
/dev/mmcblk1p1 vfat        261864  109048    152816  42% /boot
/dev/zram1     ext4         49584   22452     23548  49% /var/log
tmpfs          tmpfs        99976       0     99976   0% /run/user/1000

What you'll note there is /boot and / (root) live on two different storage devices. The EMMC is actually the / (root) device and the /boot device is local storage on the Raspberry Pi - which is actually your toolhead adapter board. The "main board" is actually a GPIO hat (sub-board) connected to the Raspberry Pi by the USB cable. When you have MCU issues because of the USB cable, it's because the Raspberry Pi lost access to the / (root) volume where all the programs are. When you flash the firmware, you are actually flashing the /boot partition with the bootloader and drivers the Raspberry Pi needs to connect to the main board.

In a more typical Voron-style design, like the BTT Manta or Octopus series boards, an actual Raspberry Pi 4 board (or BTT's version of it) attaches to the controller board via the GPIO pins - no USB cable to create a fault. While some Voron-style setups do use a toolhead board, it's a CAN bus breakout board that doesn't have any processing on it. It's just for providing a place to make connections for the heater elements, thermistors, fans, and toolhead extruder.

boydenus, 2024-06-08:
... 
> And ur right. The printer works out of the box and the update does not bring that much of a difference to the table. Also im not worried about "security" of the OS cause if u think about it,.. [...]

🙄And this is why I make six-figures restoring IT systems of companies full of people who think this way. You know you can change that password and make your system more secure? You also know just like your Wi-Fi router you are encouraged to change it, so you don't get hacked? This is why a lot of companies don't allow Wi-Fi connections to the printers and require them to be connected by Ethernet cable or USB drives only. Sure, at home it isn't a major concern - until someone walks into your house with a compromised device and it attacks your network. With Python 2.7, all it needs to be is on a network and it can be attacked via the ports you already have open on it.

McSneaky, 2024-06-12:
> Regarding power loss recovery, I spent several days testing and found that due to hardware limitations, the Z-axis of the MAX3 model cannot maintain its height during a power loss, making recovery almost impossible :(
> 
> Any solution to this issue?

If can store last known position, then could bed bottom out as low as possible and then start going back up until it reaches last known position?

Or there's no way telling, if bed has bottomed up?
I assume _should_ be able to know it, kinda like sensorless homing keeps track of steppers power usage, if it spikes, means collided with frame. Similar thing, just for bed

At least in theory, not sure, if it's possible in practise without rewriting whole Klipper..

dewi-ny-je, 2024-06-12:
> > Regarding power loss recovery, I spent several days testing and found that due to hardware limitations, the Z-axis of the MAX3 model cannot maintain its height during a power loss, making recovery almost impossible :(
> > Any solution to this issue?
> 
> If can store last known position, then after getting power back, could bed bottom out as low as possible and then start going back up until it reaches last known position?
> 
> Or there's no way telling, if bed has bottomed up? I assume _should_ be able to know it, kinda like sensorless homing keeps track of steppers power usage, if it spikes, means collided with frame. Similar thing, just for bed
> 
> At least in theory, not sure, if it's possible in practise without rewriting whole Klipper..

No it doesn't work that way, you cannot joke on the bottom of the printer is set to home at the top.

The one I'm building homes at the bottom so it can do it even with a part on the bed, and no risk of crashing the nozzle, but almost no other printer does it and the qidi(s) cannot.

End of the story...

McSneaky, 2024-06-12:
How is it not possible? Reading thro Klipper doc you can go in whichever way you want

> the axis is moved into the mechanical limit making the stepper motor lose steps. The stepper driver senses the lost steps and indicates this to the controlling MCU (Klipper) by toggling a pin. This information can be used by Klipper as end stop for the axis.

Reading some more from Klipper doc, it even states, that it works with Z axis, with Z just have to check for force applied, which might be too much for printer itself to handle, especially if hitting the nozzle

> Be sure that your mechanical components are able to handle the load ... Especially leadscrews might generate a lot of force. Homing a Z axis by bumping the nozzle into the printing surface might not be a good idea.

Qidi X3 printers don't have leadscrew attached to stepper directly, they are ran by belts, which means no so big forces will be applied. Also wouldn't be hitting the nozzle, which is real weak compared to bottom of the printer

boydenus, 2024-06-12:
Qidi declines to update the software for their printers.

closed 2024-06-12 by boydenus

dewi-ny-je, 2024-06-12:
> How is it not possible? Reading thro Klipper doc you can go in whichever way you want
>
>
> Reading some more from Klipper doc, it even states, that it works with Z axis, with Z just have to check for [...]

Look, change the firmware settings to home at bottom and try. If it works then some macros could be modified/written to save the status and resume.
If it doesn't, either it doesn't and you revert, or you have some repairs to be done.

If you are asking others to take the effort and potentially damage the printer to test something you would like, I'm afraid the chances are against.

McSneaky, 2024-06-12:
I just did test with my printer and it works all good. Modified `[homing_override]` section of printer.cfg :) Went real slow and soft, just-in-case

dewi-ny-je, 2024-06-12:
> I just did test with my printer (X-Max 3) and it works all good. Modified `[homing_override]` section of printer.cfg :) Went real slow and soft, just-in-case

Is it reliable and reproducible?
If that's the case, you could ask in the Klipper Discord to see if someone has a macro or tips to save the height reached after each layer, then at boot checks for an unfinished print and reloads the gcode from the layer specified.