QIDITECH/QIDI_MAX3 #54: Security Issue / Question - Why is my Max3 sending outbound traffic to arksine.github.io?

leckert123, 2024-06-10:
Just got my X-Max3 2 days ago, and have been printing up a storm. The machine is awesome. It is connected by Ethernet to my network.  As a routine security check, whenever new devices are added to my network, the hardware firewall traffic logs were reviewed. I like to see if anything "interesting" is going on.  Good gosh, yes.

Tons and tons of outbound traffic is being to **arksine.github.io**

1. Why? 
2. What is being sent?
3. Is that data then being routed forward to someplace else that we should know about?
4. Is merely Klipper or Moonraker checking ever few minutes to see if they are up to date? 
5. I can block that traffic...but what would happen?

I bought a Qidi, in part, to avoid the BBL backdoor data issues. Is my lovely X-Max3 doing anything nefarious? 

[image]

Fisheye3D, 2024-06-10:
Its moonraker and klipper https://github.com/Arksine

leckert123, 2024-06-10:
Thank you. I think.  No data is coming back. It's all one way. My machine is sending data to arksine, but none coming back. I'm dumb.

Fisheye3D, 2024-06-10:
> Thank you. I think. No data is coming back. It's all one way. My machine is sending data to arksine, but none coming back. I'm dumb.

The reason you see it over and over is because its moonrakers update API that qidi has blocked

leckert123, 2024-06-10:
Ah, then that's a good thing.  Because otherwise a Moonraker update would brick my machine!

leckert123, 2024-06-10:
Thank you, Fisheye3D.

closed 2024-06-10 by leckert123