QIDITECH/QIDI_MAX3 #55: X-Max3 EMMC Image files on Website are outdated and for the wrong machine

leckert123, 2024-06-21:
The EMMC image files on the QidiD.com site are WRONG and outdated.

I got the 32gb EMMC to upgrade my X-Max3.  Not knowing there were hidden partitions, I backed up the files and flashed what I assumed would be the newest image file from the main website link.  

I went to this page and grabbed the latest Google Drive download via this link that says it's for X-Max3/Plus3. 
       https://qidi3d.com/products/x-max-3-x-plus-3-x-smart-3-emmc-32g

**Well, this is not the latest image. Nor is it for the X-Max3**

1. Printer.cfg is for Plus3, and not for both.  bed_mesh settings are designed for the smaller machine, and probe offset distances are different as well.  I'm not sure what else might be screwed up.
2. There is no printer.mksini file.
3. There is no way on the machine to initiate Wi-Fi setup, which I was hoping would work when I learned about #3
4. In this older software, Fluidd does not map to the IP address, appended with :10088 like on the latest version. I could not load Fluidd to see what was going on because I was expecting to use xxx.xx.xx.xx:10088.
5. The version of the image in the Google Drive produces:
      - SOC:   2.0.06
      - MCU:  2:10.0
      - UI:      V4.3.13

In hindsight, I should have just plugged in the EMMC and been done. But I screwed up, tried to be smart and caused my own issues.

**REQUEST:**

1. Please post the EMMC image files here on Github for each machine.  I would like to be able to reflash the 32gb with the correct, current image
2. Please redirect the link for the EMMC image that's on the EMMC webpage to NEW files on Github. 

many thanks!

CChen616, 2024-06-21:
Max3 and Plus3 share the same system except for the printer.cfg file, so they use a common image. The provided image is the initial image. After receiving and installing it, you need to select your desired version, such as V4.3.13, and download the corresponding update package for use.

The update package will also include the printer.cfg file for the specific version.

Currently, we cannot provide a separate image file for each version.

leckert123, 2024-06-21:
Thank you CChen (I'm used to typing CCCChen on Discord!!)

I'm actually in the process of doing the 4.3.13 update by USB.  Would it be possible to

1. Post the current image being flashed onto EMMC's leaving the factory, even if it's the same for both machines?
2. Add wording to the EMMC web page directions explaining that one needs to replace the correct printer.cfg, following the directions here https://github.com/QIDITECH/QIDI_MAX3

CChen616, 2024-06-21:
> Thank you CChen  (I'm used to typing CCCChen on Discord!!)

😸 

> * Post the current image being flashed onto EMMC's leaving the factory, even if it's the same for both machines?
> * Add wording to the EMMC web page directions explaining that one needs to replace the correct printer.cfg, following the directions here https://github.com/QIDITECH/QIDI_MAX3

1. Our eMMC is pre-flashed with the initial image by the supplier (However, when purchasing eMMC separately, it is brand new and untouched). Before shipping from our factory, we manually update it to the latest version.

    Regarding providing image files for each version, we currently don't have enough time to do so, but it has been added to our to-do list (though the priority might be low).

2. For the 3 series machines, if there is a `printer.cfg` file in the `QD_Update` folder during the update, it will be automatically replaced. Therefore, a normal update will be enough.

closed 2024-06-21 by leckert123