QIDITECH/QIDI_MAX3 #58: Update Request: Debian Buster EOL Reached on Qidi XMAX3 Printer

justnx, 2024-07-28:
Issue:
The Qidi XMAX3 3D printer is currently running on Debian Buster OS, which has reached its End-of-Life (EOL) on June 30th, 2024. The lack of updates since EOL presents several significant risks and issues that need to be addressed urgently.

Problems Identified:

    Security Vulnerabilities:
        With Debian Buster reaching EOL, it will no longer receive security updates. This makes the system vulnerable to newly discovered security exploits, potentially compromising the printer and the network it is connected to.
    Software Incompatibility:
        Over time, software dependencies may become incompatible with newer versions of applications or libraries. This can lead to functionality issues or the inability to use new features.
    Lack of Support:
        Without official support and updates, any bugs or issues identified in the Debian Buster OS will remain unresolved. This can result in decreased reliability and performance of the printer.
    Compliance Risks:
        For users in regulated industries, using unsupported software may lead to non-compliance with industry standards and regulations, posing legal and operational risks.

Request:

    Immediate Update: We request an immediate update to the latest stable version of Debian or an alternative supported OS to mitigate the risks associated with running an EOL operating system.
    Regular Updates: Implement a policy to regularly update the OS and software dependencies to ensure ongoing security and compatibility.
    Transparency: Provide a clear update and maintenance roadmap for the OS and software environment used in the Qidi XMAX3 printer.

Steps to Reproduce:

    Check the OS version on the Qidi XMAX3 printer.
    Verify that it is running Debian Buster.
    Confirm the EOL status of Debian Buster from official Debian sources.

Expected Outcome:
A plan and timeline for upgrading the Qidi XMAX3 printer to a supported OS version, ensuring continued security and functionality.

Environment:

    Printer Model: Qidi XMAX3
    Current OS: Debian Buster
    OS EOL Date: June 30th, 2024

QIDITECH, 2024-07-29:
The system upgrade time is too long and there is a probability that the upgrade will fail, it is not suitable for the average customer, so we will not include it in the official version.
We have prepared a new version of the system, please follow the instructions to upgrade yourself if necessary.
https://github.com/CChen616/QIDI_Max3_Bookworm

trancified, 2024-07-29:
With this bookwork update, will all future Max3 updates just work? Are there any planned?

https://github.com/CChen616/QIDI_Max3_Bookworm

CChen616, 2024-07-30:
If there are updates, we will try to keep both versions synchronized, but it may take extra time to prepare.

justnx, 2024-07-31:
I'm concerned that there is no official offer from Qidi that wouldn't violate terms for after-sales service.
Wouldn't it be better to go with Bullseye? It's stable, mature, and has long-term support until August 31, 2026. 
The changes introduced in Bookworm seem to be more problematic.

QIDITECH, 2024-08-01:
More people want to update to Bookworm, and that's it.

Phil1988, 2025-01-24:
@justnx @trancified and @whb0514 
[**FreeDi**](https://github.com/Phil1988/FreeDi) uses recent bookworm.