QIDITECH/QIDI_MAX3 #61: Max3 4.3.15 - virus detected Misleading:Linux/FRP.B!MTB

trancified, 2024-09-29:
Just tried to download the latest Max3 firmware update 4.3.15, and Windows defender is saying there is a virus: Misleading:Linux/FRP.B!MTB. Please fix it.

leckert123, 2024-09-29:
I downloaded it last night immediately after it was released and installed it. Norton 360 was perfectly happy with it, even when manually checked.  It's fine. 

Windows Defender loves to provide false positives when it doesn't have history of a file that's not been downloaded thousands of times.  WD is the problem, not the firmware package.

trancified, 2024-09-29:
Normally I would agree, however, this wasn't just a "we don't know about this" kind of error, it specially called out an actual issue. And no offense, but I wouldn't trust Norton if it were the last virus scanner on earth.

leckert123, 2024-09-29:
No offense taken.  But I've cleared my other device firmware updates through multiple scanners and honestly, the only ones with false positive issues are WD and BitDefender.  

My printer is happily running the new firmware. It's printing right now and was printing minutes after doing the update via flash drive. 

Here's 3rd party AV testing. Enjoy.  
https://www.av-test.org/en/antivirus/home-windows/

trancified, 2024-09-29:
I think Qidi should confirm, and if clean, submit the firmware directly to Microsoft via their submission process to clear the error. I mean, I am just one person, and with supply chain injection issues on the rise, we can't be too careful.

QIDITECH, 2024-09-30:
We use frp for some data communication. Because it has the function of transmitting data, it is always identified as a virus by security software. But rest assured, he doesn't have any malicious features.
You can learn more by following the links below:
https://github.com/fatedier/frp/issues/2095
If you still feel that it is not secure, please do not update the firmware.

Fisheye3D, 2025-01-09:
Its not a virus, Its a broken linux protocol that shouldn't be used but its not a virus

[image]