QIDITECH/QIDI_MAX3 #62: Max3_V4.3.15 wiped out TimeLapse functionality

leckert123, 2024-10-04:
Max3_V4.3.15 appears to have wiped out timelapse functionality  Timelapse no longer appears in Fluidd. Slicer profiles that included timelase g-code no longer generate timelapse files.  The timelapse icon in the sidebar of Fluidd is gone.

I have re-uploaded the timelapse tar, cfg, and sh files and ran the bash command per the instruction guide.  An error is shown after the bash command.  I even downloaded them again to see if these were updated and re-uploaded them.  After Klipper was restarted, timelapse still does not show up in Fluidd.

Klipper now won't connect to the printer after restarting Klipper as well as doing a restart of the machine itself.  I can see the camera in Fluidd, so the IP address, which is fixed, is fine.  Moba still connects 

In the future, please don't issue firmware updates just a couple of days before going on a long national holiday. Tonight I'm going to pull the eMMC and install my backup eMMC with the prior firmware version until you can get this system fixed.

Guide followed:
https://drive.google.com/drive/folders/1jSElfWv5rAiQaPWIu1H7EB9CT8y3OOml

Attached shows the status and files. 

[image]
[image]

leckert123, 2024-10-04:
I just learned that the remote restart in Klipper does not work.  In fact, it produces an error on the machine

I did a hard power cycle restart at the machine, on-site.  It rebooted with an error about starting abnormally, but then gave me a normal screen.  Then this message:

[image]

Went home, power cycled the machine, and it seems fine. 

So final comments on what's wrong:

1. Latest firmware wipes out timelapse from the Klipper installation
2. Reinstalling timelapse files requires manual navigation to the folder to ensure they are correctly uploaded.
3. After successfully running the bash command, the machine needs to be power cycled manually.  Restarting Klipper does not work by itself.
4. The Host Reboot option in Klipper does not actually reboot the machine the same way a power cycle does.  In fact, it caused Klipper errors. The machine requires a hard power cycle to reboot without error.

[image]

QIDITECH, 2024-10-05:
Yes, changes to the klipper require a power restart. Thank you for your feedback and sharing.

leckert123, 2024-10-05:
Thanks for listening.  Hopefully there will be a fix possible to prevent people from having to re-set-up timelapse after doing the firmware update.  And meanwhile, it would be good to add a notification on the download page stating that timelapse will need to be reinstalled, with a link pointing to directions on how top do it.