QIDITECH/QIDI_MAX3 #63: x-max3 firmware  V4.3.15 error with endstop Z

corra84, 2024-10-06:
Hi everyone, this morning I installed the new firmware but as soon as I tried to print I get this error: gcode error:!! Endstop z still triggered after retract.

Has this happened to anyone??

My x-max3 have a BL touch not the probe maybe can this depends for that??

thanks for help.

QIDITECH, 2024-10-07:
I think it is caused by you using the parameters of the probe, please re-import the parameters of BLtouch.
For details, see the home page.

corra84, 2024-10-08:
> I think it is caused by you using the parameters of the probe, please re-import the parameters of BLtouch. For details, see the home page.

Yes I make a mistake because I update the firmware without delete the parameters of probe so I can try to re-update the firmware just with the BL-touch setting. You think I can update the same firmware just with different parameters??

Fisheye3D, 2024-10-08:
> Yes I make a mistake because I update the firmware without delete the parameters of probe so I can try to re-update the firmware just with the BL-touch setting. You think I can update the same [...]

You don't need to re update any firmware, Just add bl touch settings to your klipper config. Just change out the Probe section `[bltouch]
sensor_pin:^MKS_THR:gpio21
control_pin:MKS_THR:gpio11
stow_on_each_sample: False
x_offset: 28
y_offset: 4.4
z_offset: 0.0
speed: 5
samples: 3
samples_result: average
sample_retract_dist: 3.0
samples_tolerance: 0.08
samples_tolerance_retries:3`

corra84, 2024-10-09:
Ok I need just write the string on klipper co figuration in the first page on fluidd right???

wirlybird, 2024-10-09:
Just ran into this.  Follow the directions for version 4.3.13. for renaming cfg file.
In the folder with update files.
Remove the CFG you don't need such as BL Touch and rename the other CFG in the update folder to "printer".
Rerun the firmware update and you will be fine.  It will finish without any more prompts.

QIDITECH, 2024-10-10:
You just need to rename the cfg file and replace this file on fluidd and restart the firmware.
[image]

closed 2024-10-10 by QIDITECH

corra84, 2024-10-26:
Thanks I solved the problem just this morning because I don't have the printers with me.