QIDITECH/QIDI_MAX3 #66: White screen data error

capnkirk643, 2024-11-10:
Unpackaged 2 days ago plugged in screen boot up . Message on it "data error" unable to go any further. Total piece of crap. Printer looks like it was sold and returned once before. Dont recommend i will stick with creality

QIDITECH, 2024-11-11:
We're sorry to hear about the issue with your printer.
Violent transportation may cause partial damage to the internal packaging, but it is definitely not a second-hand product, please rest assured to use.
The "data error" message can often be resolved by performing a recovery via USB following the instructions provided in our [guide](https://github.com/QIDITECH/QIDI_MAX3).
If you need further assistance, please don't hesitate to reach out to our support team. Our customer support is available to help and will respond within 24 hours.
Thank you for your patience, and we hope to assist you in getting your printer up and running smoothly!

QIDITECH, 2024-11-12:
I am very sorry, please contact after-sales service to explain your situation, we will replace the motherboard and screen for you.

capnkirk643, 2024-11-20:
So i tried the work around with the file that yall sent but still nothing.
And the emmc that i ordered does nothing for it. I still think it is the
motherboard or screen so please send replacements so i can enjoy this bad
boy and make some beauty

QIDITECH, 2024-11-21:
I am very sorry, I have reported the situation to the after-sales service, but because your email address is inconsistent with your user name on github, we can't tell which one is you. Please send the page link to the after-sales service, so that he accurately understand your situation, and then send the screen board to you.

capnkirk643, 2024-11-21:
Not sure what page link you are referring to but i dont believe it to be
the screen. I think it is the main board

QIDITECH, 2024-11-22:
What I mean is to let the after-sales service know our conversation, so that he can solve your problem more accurately and quickly.