QIDITECH/QIDI_MAX3 #69: Camere Disapears

cptcl, 2024-11-26:
Hello
I have this problem for a while but, now that i do longer prints i rly miss it.
So my X-Max 3 is on https://github.com/QIDITECH/QIDI_MAX3/releases/tag/Max3_V4.3.15 @32gbmmc firmware and prints well.
But...
it constantly looses the cam after a few minutes.
Using the original Qidi Cam. Install as intended. No stress or hard bends on the USB cable.
What i do...
If my Printer is not powered on and i start it up, everything seems good and normal. I can see the camera working.
It does not matter if i start a print or just leave the printer running in idle..
After a few minutes (like 5-15) my cam just disapears, and i cant get it back working except for unplug printer, wait a minute and replug the printer. 
Event restarting the Webdamd service does not help.

To clear things up more...
My hotend adapterplate died recently, so i ordered a replacment, installed it and then i reseted the whole printer (yes, Display, Hotend, Mainboard and the mmc) to stock, than updated to 4.3.15 firmware
I even had this problem on 4.3.13

This is what i see after a few minutes
[image]

*Asking about config file...? its stock.

I hope someone can help me.
My guess is, that it has to do with camera detection,.. 

" ls -l /dev/video* "
gave me

crw-rw---- 1 root video 81, 0 Nov 25 12:17 /dev/video0
crw-rw---- 1 root video 81, 1 Nov 25 12:17 /dev/video1
crw-rw---- 1 root video 81, 2 Nov 25 12:17 /dev/video2
crw-rw---- 1 root video 81, 3 Nov 25 12:17 /dev/video3
crw-rw---- 1 root video 81, 5 Nov 25 15:17 /dev/video5
crw-rw---- 1 root video 81, 6 Nov 25 15:17 /dev/video6

a fast " usb-devices "
gave me following

T:  Bus=02 Lev=02 Prnt=02 Port=02 Cnt=01 Dev#=  4 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1bcf ProdID=0b09 Rev=10.20
S:  Manufacturer=SYX-231020-J
S:  Product=HD Camera
C:  #Ifs= 2 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#=0x1 Alt= 0 #EPs= 0 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo

Long Version with " lsusb | grep -E -i 'camera|webcam|video' "

<details>
  <summary>Long text</summary>
  
  
  ```java
Bus 002 Device 004: ID 1bcf:0b09 Sunplus Innovation Technology Inc.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1bcf Sunplus Innovation Technology Inc.
  idProduct          0x0b09
  bcdDevice           10.20
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
[... 529 lines cut ...]
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x13fc  3x 1020 bytes
        bInterval               1

  ```
  
</details>

PS: When i had this problem on 4.3.13 i did a switch in the config from 
camera_usb_options="-d /dev/video4 -r 640x480 -f 10"
to something like (i changed numbers and restarted webcamd service until i got the cam back, cant remember if it was 5,3,2 or 1)
camera_usb_options="-d /dev/video3 -r 640x480 -f 10"
and than it worked for some time but ended up in the same problem anyways

Hope someone knows how i can fix that

QIDITECH, 2024-11-27:
Please contact after-sales service, they will help you identify the problem and resolve it.

cptcl, 2024-11-29:
Sent a mail to customer service. Will update if i got a solution.

cptcl, 2024-12-23:
Solved.
Got a new camera from qidi support. This fixed the problem. Strange casue the old cam does work day long on my windows PC, but will disappear within minutes on my x-max 3.

closed 2024-12-23 by cptcl