QIDITECH/QIDI_MAX3 #70: #61 and #68 does not help me

WhatFreshHellAwaits, 2024-12-13:
I just barely got this printer about a week ago. Now I'm getting this error. All I was doing was following the on screen directions for changing out the filament. I emailed in to support, but found this site after. So trying both for now. This error will not even let me do a factory reset, nor will it pick up the USB in an attempt to reload the firmware. None of the commands on the other post will successfully run either.

QIDI X-MAX3

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.11.0-6-g336cc92a
MCU(s) which should be updated:
mcu: Current version v0.10.0-530-g3387a9c2-dirty-20221121_013639-mkspi
MKS_THR: Current version v0.10.0-530-g3387a9c2-dirty-20221210_215217-mkspi
Up-to-date MCU(s):

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'MKS_THR': Unable to extract params from: query_thermocouple

leckert123, 2024-12-13:
The toolhead and mainboard can no longer communicate with one another.

This typically happens when one has pushed the update button prompt in the Fluidd interface. That Fluidd update capability was disabled in more recent firmware versions earlier this year.  QIDI removed the update notification to avoid customers being prompted to update to incompatible firmware.  If you did the update, you'll need QIDI to rescue you.

WhatFreshHellAwaits, 2024-12-13:
> The toolhead and mainboard can no longer communicate with one another.
>
> This typically happens when one has pushed the update button prompt in the Fluidd interface. That Fluidd update capability was [...]

Awsome... J/K... I don't think I pushed that but I've done dumber things... I was just changing filament when it happened lol.. Thanks for the reply.

WhatFreshHellAwaits, 2024-12-13:
Yeah it's a brand new machine. I'm not by printer anymore but pretty sure it says its the .10 version on the screen. Only that error says .11. It's pretty weird.

leckert123, 2024-12-13:
The email support (AMS is After Market Support) will be able to assist, if they are not working with you already. I usually got answers within 12-24 hrs for my Max3 from Linda at max3support@qd3dprinter.com.  Honestly, they are excellent at trying to help, and you are definitely not the first person with this issue.

https://qidi3d.com/pages/warranty-policy-after-sales-support

Max3support@qd3Dprinter.com
Max3AMS@qd3Dprinter.com

WhatFreshHellAwaits, 2024-12-13:
They replied with instructions... a little SSH and deleting Klipper, uploading Klipper form GitHub, deleting Moonraker, uploading the replacement from Github, then updating to 4.3.15 and printer no longer appears to be a brick...  very interesting lol... I'm not sure why if I'm using the QIDI slicer only, with the QIDI printer, that it would even prompt for any update, or appear to update automatically to what they are calling a 3rd party Klipper software. Not bad response time at 7 hours... that's quicker then some venders reply in my job. Thanks for sharing the above information.

leckert123, 2024-12-13:
Very happy it worked out. QIDI's customer support is the best I've witnessed in this industry.  They truly try to help and are concerned about customers.

The "update" prompt I mention is not in QIDI slicer. It appeared at the top right of the Fluidd interface page that you see in your web browser. You get this interface when you type in the IP address for your machine, plus 8080 at the end for the port.  

Again, congratulations that you are back up and running. Enjoy printing!  It's an amazing machine. I just put up another print on mine.

WhatFreshHellAwaits, 2024-12-13:
> Very happy it worked out. QIDI's customer support is the best I've witnessed in this industry. They truly try to help and are concerned about customers.
>
> The "update" prompt I mention is not in QIDI [...]

It was updating to the new 4.3.15 when I wrote that... after update I got a whole new surprise... "The System Starts Abnormally!" Waiting on their reply on this one now... I really don't have the patience for this on a week old device lol...

QIDITECH, 2024-12-14:
I'm very sorry to let you meet so many problems, I have contacted the after-sales service, they will help you solve the problem as soon as possible.

WhatFreshHellAwaits, 2024-12-14:
> I'm very sorry to let you meet so many problems, I have contacted the after-sales service, they will help you solve the problem as soon as possible.

I have an email in with them right now. However, there is not ticket number to reference the case.

arcangelraider, 2025-03-09:
> They replied with instructions... a little SSH and deleting Klipper, uploading Klipper form GitHub, deleting Moonraker, uploading the replacement from Github, then updating to 4.3.15 and printer no [...]

Hello. I have the same issue, after preesing the update on Fluidd.  Was Qidi able to resolve the problem on you printer?