QIDITECH/QIDI_MAX3 #71: Sound button

KazakovAndrey, 2024-12-28:
Hi!
The sound button always reverts to "off" after you power down. Is there a way to save it on?

QIDITECH, 2024-12-30:
You can do this by modifying the value in print.cfg in fluidd.
[image]

KazakovAndrey, 2024-12-30:
This doesn't solve the problem. 
This section exist in print.cfg by default from firmware.

QIDITECH, 2024-12-31:
After changing the value from 0 to 1, save and restart the printer.

KazakovAndrey, 2024-12-31:
I try, but nothing has changed.

look cosole after start and see beep_off
[image]

QIDITECH, 2025-01-02:
You need to change the value of output_pin sound, not beeper.
Or you can send your print.cfg file to me, and I will modify it and send it to you.