QIDITECH/QIDI_MAX3 #72: Re-write MotherBoard

Ferropluss, 2025-01-19:
Hello. Tell me, is there an update only for the motherboard? When there is no EMMS.
Via USB or TF?
The problem is that there is no wired connection.
MKS X-6 V 1.0

QIDITECH, 2025-01-20:
I'm sorry, I don't understand what you mean. If you want to reset the program, just refresh emmc.

Ferropluss, 2025-01-20:
Hello. What does x-4.bin firmware mean? Then it is possible to update the motherboard system? Or I'm asking the wrong questions. Do you have any support for restoring an Ethernet connection?

QIDITECH, 2025-01-21:
sorry, If you do some work on the motherboard and firmware yourself, we can't give a clear answer.

Ferropluss, 2025-01-21:
I don’t do it myself. I am interested and want to understand what affects the lack of connection via IP cable.

Ferropluss, 2025-01-21:
I found the reason, but I can’t completely cope with it. Static address 169.254.255.255 and no from the Dynamic printer. LAN connection is not possible. Maybe there is a manual setting for Qidi x-max3?

QIDITECH, 2025-01-23:
The IP is assigned by the router, and if the interface is not faulty, it can only be caused by the router's Settings.

Ferropluss, 2025-01-23:
The reason is in the firmware, since after the update the klipper (mcu) settings changed. managed to fix klipper.uf2 and x-4.bin . Help other users understand and configure this. Your support is important to us. Please attach files for firmware.