QIDITECH/QIDI_MAX3 #74: Where is z-offset stored

richardklingler, 2025-01-29:
Adjusting the z-offset has a sort of backslash problem.

Let‘s say at 0.9mm you feel a strongresistance when moving the supplied sheet under the nozzle.

You have then go at least back to 0.8mm until a slight resitance is felt.

But this causes the first layer to be printed too thin.

Now where is the saved z-offset stored sonit can be manually edited?

Normally it is stored in printer.cfg, but not with this printer.

QIDITECH, 2025-01-30:
Check the save_variables.cfg file.

richardklingler, 2025-01-30:
There is no such file when I ssh into the printer and search for it with:

`find ./ -type f -name "save_variables.cfg"
`

richardklingler, 2025-02-01:
Only thing I've found is in
`klipper_config/config.mksini
`
Where a value represents the current z-offset:

```
[babystep]
value                          = -1.010000

```

The supplied testing paper has a thickness of 0.25mm...so is also the first layer where I have always gaps in the infill.

So if I want to move the nozzle closer just set the babystep value to a lower negative value?
For example if I want it to be 0.05mm closer than it is now, just set it to:

```
[babystep]
value                          = -1.060000

```

Can someone confirm this?

QIDITECH, 2025-02-03:
You can view and edit the file directly in fluidd without ssl connection.

Explore2Create, 2025-08-09:
> You can view and edit the file directly in fluidd without ssl connection.

what does this response mean? Qidi, can you just answer questions? Why does it feel like pulling teeth when we ask you questions. This really shouldn't be this complicated. I just don't get it.