QIDITECH/QIDI_MAX3 #75: z offset 1.230mm now probe movement doesn't trigger at the end of auto bed leveling

xenonreality, 2025-01-31:
Recently got a creality volcano k1 nozzle & used a diamond saw to cut it to length. But after adjusting the z offset in the auto bed leveling program it says probe movement won't trigger. it seems like when it starts going back down to the bed.

I have already had it crash into the bed several times trying to get it to go back to z home as well.

What setting do I need to change in the printer.cfg file to be able to change it to work properly?

I don't want to damage anything anymore than it already has been after it kept going into the bed & skipping off the z-screw rails until I could get to the switch in the back.

I decided, instead, to just take the nozzle out then run the auto bed leveling program then put the nozzle in & set the z-offset. Now it can't handle the probe movement at the end. Every time it makes a huge racket again then stops & tells me.

xenonreality, 2025-01-31:
I have changed the probe to -6 offset & so, so many other things besides this. It never slows down & goes straight to through the bed. Maybe I have a defective sensor?

QIDITECH, 2025-02-03:
Please contact customer service to replace the sensor.

xenonreality, 2025-02-04:
I have contacted customer service as well. During the firmware update I made a mistake & used the printer.cfg file that is correct for the one I have but I forgot to delete the part that specifies the sensor type. It still compiled & booted up just fine. Would that make the sensor no longer work? As I have seen online reports of reverting firmware back can cause the sensor to not work & updating from there also doesn't fix it.

Now I know that can be used error as well, but if a firmware update, that is basically a bios flash, can actually brick functionality of the device & is then no longer fixable but it looks like a virus to your virus protection & it needs to have the title changed to work (if that's the case), while also being no where stated except the previous firmware update page that has it in a small unbolded, not outlined, & not heavily pictured area that someone would easily just skip through & most users who buy your devices aren't willing to even look at the "scary" readme file, download the "virus" & or even want to put up with a terminal in anyways (if it was ever necessary) then I can see why people seem to always need to rely on customer service support.

I am, genuinely, grateful that you do send parts & are often very helpful. But if the company is survived by customer service support then it becomes incredibly hard to be able to recommend your product to others.  The user experiences should be understood better than this, the device couldn't connect to your phone without an account & an update. The only way to auto update is with Internet functionality, which basically isn't there until then. If you want to go back & forth with a flashdrive for it to work I guess that's okay, but a rear facing Ethernet cable & if bed crash happens the only switch is hard to reach back there & there is nothing on the touch screen it really is a terrifying user experience!

A lot of this printer seems great! In fact, if the sensor only needs a magnetic plate thing you sell, then it should be there from the start. If you have updated firmware, its worth it to unbox & update it to the latest version because users will not freak out, a firmware update that makes a white screen with red lettering is also scary to users who don't know what it means & that it's just compiling. Which if the .cfg file having the title wrong is capable of allowing some functionally still but hides the sensor & or absolute values from be changed but still records the values with the sensor but doesn't use them because of that title being wrong that's a huge problem! 

That's if the problem even is that, if. Reverting firmware back can cause the same issue. Apparently. Which now makes me worried to even try it. If it's not that then it's sensor & or plate. Which...what it's got strong magnets & seems like that shouldn't be needed. If the community like the touch sensor but you are recommending the inductive it's because it's not accurate & or often doesn't work without them buying something that should just be included for it to even work. Because then the code is not actually working, it's working off of hard values that can't be altered with the way it's coded.

It's scary & a user experience that will absolutely drive any customers away from your product. There is a growing segment of people that are grandparents who have time & money & know enough about stuff like this (I know a lovely lady who has a printer & she's in her 60s she printed out a segmented dragon & it looked nice & I live in a rural area with not a big tech community in the slightest) that want to buy something like this to play around with & enjoy, let alone non-enthusiasts (I'm an enthusiast) who also wants to buy something & this would easily make them not want to. The volatiles & more are a huge reason to have an enclosure & the larger print volume with it being made of mostly metal this is a very good foundation printer. But stuff like this destroys peoples desire to even look even if customer service is good. 

I send another email to them and see what I should try doing moving forward.

closed 2025-02-04 by xenonreality