QIDITECH/QIDI_MAX3 #79: Problem after net firmware installed  4.3.15 with my QIDI X-MAX 3

Liorkatz, 2025-03-30:
hi there i just finish update my printer QIDI x-max 3 and i got stuck .. :( 

I downloaded the update copy to the SD plug in the back and then go to update and press update 
it's says that i need to reboot so i did' after that I numbers were got out and looks like the real deal .. 

after first reboot 

at start its saying that its haveing problems with the Z  
 " !! Endstop z still triggerd after retract" 

i have tryied few times to start its move to the right side to front and the bad moving down a few and stuck again 

its says he wanted to restart so i did' unplug it and now this stuck! 

it wound even start the welcome screen its saying
"THE SYSTEM STARTS ABNORMALLY!  please turn off the machine and wait a for a while before restarting "

after getting stuck i just saw that I was needed to change the file name ... ggggg 
I know I needed to read it before but ...  i didn't 

what can i do ?  plz help

QIDITECH, 2025-03-31:
Make sure that the configuration you updated (printer.cfg file) is correct.

Liorkatz, 2025-03-31:
> Make sure that the configuration you updated (printer.cfg file) is correct.

I wish I could do this ..
It's stuck on start up

I can't do anything

QIDITECH, 2025-04-05:
I am very sorry, please contact the after-sales service and describe your problem in detail, they will help you solve the problem.