QIDITECH/QIDI_MAX3 #80: THE SYSTEM START ABNORMALLY after firmware update.

mauser348, 2025-04-05:
Verified right probe.

changed name of file did the update to firmware now I get this error "THE SYSTEM START ABNORMALLY"

[image]
[image]

QIDITECH, 2025-04-10:
I am very sorry, please contact the after-sales service, they will help you solve the problem.

sambathrajv, 2025-10-08:
I am having same issue. Reached out to the support team. Something wrong with this version. I was forced to do a firmware update since the QIDI software showed me that updates for the klipper was available and after installing them, it showed me version mismatch with the firmware. I thought using this firmware will solve the issue. But completely bricked my printer.